Qucs/qucs #934: Qucs examples: Suggestion for improvement

andresmmera, 2019-06-08:
The set of examples is a bit chaotic: circuits are not classified by topic and many of them have confusing names (such as bbv.sch, gain.sch, charac.sch, etc.). Additionally, the schematics don't contain graphs nor data display windows. Often, text and components overlap in the schematics. In this sense, I would like to help putting some order there as well as adding some circuits to the example set.

From my point of view, the first step is to group the schematics by area and then create a separate project for each topic. The directory tree should look like this:

Examples
├── DC
│   ├── Current sources_prj (Empty. I can put some stuff here)
│   ├── DC converters_prj
│   │   ├── boostconverter.sch
│   │   ├── buckboost.sch
│   │   └── buckconverter.sch
│   ├── DC_General_prj
│   │   ├── bridge.sch
│   │   ├── diff1.sch
│   │   ├── Puls3b.sch
│   │   ├── schmitt.sch
│   │   └── time_resistor.sch
│   ├── Operational amplifiers_prj
│   │   ├── gain.sch
│   │   ├── opamp_gyrator.sch
│   │   └── singleOPV.sch
│   ├── Rectifiers_prj
│   │   ├── fullwaverectifier_1.sch
│   │   ├── fullwaverectifier_2.sch
│   │   └── supply.sch
│   ├── Voltage regulators_prj
│   │   └── preregulator.sch
│   └── Waveform generators_prj
│       ├── sawtooth-1.sch
│       ├── sawtooth-2.sch
│       ├── sawtooth-3.sch
│       └── sawtooth-discreet.sch
└── RFMW
    ├── Active filters_prj
    │   ├── active_bp.sch
    │   └── active_lp.sch
    ├── Amplifiers_prj
    │   ├── Amp_Baseband_Low-noise.sch
    │   ├── Amp_Two_Tone.sch
    │   ├── bbv.sch
    │   ├── fet_noise.sch
    │   ├── selective_amp.sch
    │   └── stab.sch
    ├── Balun_prj
    │   └── IdealBalun.sch
    ├── Couplers_prj
    │   ├── mscoupler.sch
    ├── Combiners_prj
    │   └── wilkinson.sch
    ├── Filters_prj
    │   ├── bpf_10Ghz.sch
    │   ├── BPF_1550_edge_cpld.sch
    │   ├── BPF_2000_Co-ax.sch
    │   ├── chebyshev1_5th.sch
    │   ├── elliptic_5th.sch
    │   ├── filter_optimization.sch
    │   ├── LPF_1000_Coplanar.sch
    │   ├── LPF_1000_Radial-Stub.sch
    │   ├── LPF_1250_Elliptical.sch
    │   ├── LPF-Balun2.sch
    │   ├── LPF-Balun3.sch
    │   ├── notch.sch
    │   └── resonance.sch
    ├── Frequency multipliers_prj
    │   ├── multiplier.sch
    │   ├── X2_100_Bipolar.sch
    │   └── X5_500.sch
    ├── Mixers_prj
    │   ├── active_mixer.sch
    │   ├── gilbert.sch
    │   ├── mixer.sch
    │   └── single_balanced.sch
    ├── Non-linear_prj
    │   └── diode_hb.sch
    ├── Oscillators_prj
    │   ├── classic_osci.sch
    │   ├── colpitts_base.sch
    │   ├── colpitts.sch
    │   ├── lc_osc.sch
    │   ├── lf_osci.sch
    │   ├── rf_osci.sch
    │   ├── sym_osci.sch
    │   └── VCO_100.sch
    └── RF_General_prj
        ├── charac.sch
        ├── chargepump.sch
        ├── Comb_Gen.sch
        ├── fet.sch
        ├── fgummel.sch
        ├── fhr01fh.sch
        ├── flicker.sch
        ├── giacoletto.sch
        ├── groupdelay_ac.sch
        ├── groupdelay_sp.sch
        ├── gyrator.sch
        ├── microstrip.sch
        ├── pentode.sch
        ├── rgummel.sch
        ├── shot.sch
        ├── SmithChartTest.sch
        ├── swr_meter.sch
        └── triode.sch

and the Qucs navigator would look like this:

[image]

Do you agree with this?

in3otd, 2019-06-08:
yes, please! The examples surely need some refreshing. I would even go slightly further and make every example a separate project, i would make things easier when adding more complex examples.
They are in a single directory mainly for historical reasons - there was no project navigation until recently...

guitorri, 2019-06-09:
I think that most examples are already in qucs-test repo as individual
projects. Maybe we should pull from there? The examples should be tested
anyway.

felix-salfelder, 2025-01-18:
We have started to use examples from qucs-test as tests. I think each directory is actually a "project".
I do not see the need for the/a qucs-tests repo. Detached from the code, it just adds work.

Now we could rename the "tests" directory to "examples". We could as well tag and install a subset of the tests as examples.