Qucs/qucs #936: broken parse_result.py in 0.0.19

gsteele13, 2019-06-11:
I've noticed that parse_result.py gives an error when trying to parse the .dat xml files (error about casting to an int)

It also ends up producing dependent variable arrays with multiple empty dimensions

I found a simple fix. I'm not sure what the procedure is for submitting a fix officially: in any case, it's pretty easy by adding astype and squeeze:

```
    data['variables'] = variables
    # reverse the shape variable in order to get the reshape operation (see below)
    # correct
    shape = shape[::-1].astype('int')
    shape = shape.squeeze()
```

and a squeeze here:

```
            data[key] = np.reshape(temp_data, shape).squeeze()
```

cheers,
gary

felix-salfelder, 2019-06-12:
disclaimer: i have never used this script, nor do I intend to do so.

On Mon, Jun 10, 2019 at 07:52:24PM -0700, Gary Steele wrote:
> I've noticed that parse_result.py gives an error when trying to parse
> the .dat xml files (error about casting to an int)

is this a regression? a bug in the xml generator? (qucsator?) can you
give an example or a reference to an example?

> I found a simple fix. I'm not sure what the procedure is for
> submitting a fix officially: in any case, it's pretty easy by adding
> astype and squeeze:

thanks for the hint!

anybody can email a patch or a git commit. you could file a merge
request on the website. The difficult bit will be to explain what it
does, and why it is needed (easier with a test case or a test).

I found that the script does not have a header. perhaps it should be
added to state license & author, maybe purpose & usage? (not mandatory).

gsteele13, 2019-06-12:
Here is the error I get (sorry for not posting earlier, had already fixed it and needed to re-download):

```
In [3]: parse_result.parse_file("3d_cavity_amplifier.dat")                      
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
/anaconda3/lib/python3.7/site-packages/numpy/core/fromnumeric.py in _wrapfunc(obj, method, *args, **kwds)
     50     try:
---> 51         return getattr(obj, method)(*args, **kwds)
     52 

TypeError: 'numpy.float64' object cannot be interpreted as an integer

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
<ipython-input-3-ef7c3961c6c1> in <module>
----> 1 parse_result.parse_file("3d_cavity_amplifier.dat")

~/SteeleLab/projects/Gary/qucs/parse_result.py in parse_file(name)
     76         if temp == 'dep':
     77             temp_data = data[key]
---> 78             data[key] = np.reshape(temp_data, shape)
     79 
     80     return data

/anaconda3/lib/python3.7/site-packages/numpy/core/fromnumeric.py in reshape(a, newshape, order)
    277            [5, 6]])
    278     """
--> 279     return _wrapfunc(a, 'reshape', newshape, order=order)
    280 
    281 

/anaconda3/lib/python3.7/site-packages/numpy/core/fromnumeric.py in _wrapfunc(obj, method, *args, **kwds)
     59     # a downstream library like 'pandas'.
     60     except (AttributeError, TypeError):
---> 61         return _wrapit(obj, method, *args, **kwds)
     62 
     63 

/anaconda3/lib/python3.7/site-packages/numpy/core/fromnumeric.py in _wrapit(obj, method, *args, **kwds)
     39     except AttributeError:
     40         wrap = None
---> 41     result = getattr(asarray(obj), method)(*args, **kwds)
     42     if wrap:
     43         if not isinstance(result, mu.ndarray):

TypeError: 'numpy.float64' object cannot be interpreted as an integer
```

The schmatic file is a simple one that includes a single S-parameter sweep (see attached, it is an .sch file but I had to rename it as a .txt)

I'll look at sending a patch

Cheers,
Gary

gsteele13, 2019-06-12:
Sorry, missed file

[3d_cavity_amplifier.txt](https://github.com/Qucs/qucs/files/3281438/3d_cavity_amplifier.txt)

gsteele13, 2019-06-12:
(it is actually a .sch file but I had to rename it to .txt to be able to upload it in this discussion thread)

closed 2020-02-19 by guitorri