Qucs/qucs #938: [Feature Request] Add exporting to KiCAD schematic and footprint

easyw, 2019-10-17:
Hi, I noticed this two projects:

- [Qucs-RFlayout](https://github.com/thomaslepoix/Qucs-RFlayout)
- [qucs2kicad](https://github.com/Valber/qucs2kicad)

Is there any chance to integrate those projects into a Qucs to KiCad exporter?

thomaslepoix, 2019-11-18:
Hi, thank you!

About Qucs-RFlayout, a discussion started [in the mailing list](https://sourceforge.net/p/qucs/mailman/message/36768803/). I would like this project to be little more mature before starting to integrate it.

felix-salfelder, 2021-02-21:
> Is there any chance to integrate those projects into an integrated Qucs to KiCad exporter?

Schematic file formats are plugins in future Qucs. There is plenty of space for new plugins, and support for several formats is intended.

"Export" is pretty simple, and a good starting point. Take and modify an existing one (e.g. Verilog), or implement it from scratch.

closed 2025-03-02 by felix-salfelder

thomaslepoix, 2025-03-06:
Also, Qucs-RFlayout get integrated to Qucs-S as of [24.2.0](https://github.com/ra3xdh/qucs_s/releases/tag/24.2.0) version. (https://github.com/ra3xdh/qucs_s/issues/465)