Qucs/qucs #941: Source will not compile on Qt 5 and OpenSuse rpm builds flawed.

ajohnw, 2019-12-14:
Any attempt to compile any of the versions I can find results in an error message
_Found unsuitable Qt version "" from NOTFOUND, this code requires Qt 4.x_
but I also seen > than 4.6 on other versions.

There are 2 rpm's for Opensuse 15.1. Schematics work well on both version 0.18 and 0.19. As soon as simulate is selected this error is reported
_/usr/bin/qucsator: symbol lookup error: /usr/bin/qucsator: undefined symbol: redirect_status_to_stdout_
The netlist does seem to be generated. Qt4 has been obsolete for some time now. This missing symbol may be all that is wrong.

A minor point on schematics. The R,C etc symbols set what the items are so there shouldn't be any need to display R= and C= etc. Their addition makes the production of neat schematics more difficult. It does seem to be possible to make the notations appear above, below etc according to orientation. That does help.

John
-

ajohnw, 2019-12-14:
Managed to get further. Qt5 on OpenSuse 15.1 comes with some Qt4 libraries. Installing the devs to go with that and qucs will compile. Rather a lot of depreciated warnings though. Many looked to relate to threads. It took a long long time so I went out.

An example appears to work now but my schematic which uses the audio freq range doesn't. I get this error message

_line 72: syntax error, unexpected Identifier, expecting '"'_

I could do with a clue on what might cause that. I have moved items about to tidy up the schematic and on a couple of items I had to delete to get at traces of wiring that was left and then put them back. Also not clear on labelling syntax. For instance is a resistor marked as 5.1k ok or does it need to be 5.1 k with the space or etc.

The same appears to apply to Qucs-S which I am not keen on using unless I can find a lot of analogue devices to install for ngspice. What comes with Qucs suites me at the moment and I understand others can be added.

John
-

in3otd, 2019-12-16:
The current Qucs code compiles only with Qt4.

I have no idea who prepared the rpms for Opensuse, the current 'master' and `develop` branches compile cleanly on Ubuntu and Windows.

We are aware Qt4 is obsolete and there was/is some work going one since a lot of time to port Qucs to Qt5; there are a few discussions about this in the Issues and Pull Requests here on GitHub and on the qucs-devel mailing list.
The problem is that only a couple of people were/are working on that and apparently they did not have much spare time lately.
If you want to play with what has been done so far for Qt5 take a look at these branches
https://github.com/Qucs/qucs/tree/develop+qt5-10
https://github.com/Qucs/qucs/tree/qt5-6-merge-rc1
and check their respective `.travis.yml` files to see how to build the code there.
Note that the code there is surely still somewhat buggy/unstable/incomplete, AFAIR.

Unfortunately there is no easy way to link the simulator errors as you got above to a component on the schematic currently. You should manually open the last netlist (press F6) and check which component is mentioned in line 72 of the netlist.

Regarding the "neat schematics", in short it's all about personal preferences. The fact that Qucs components have always been shown in that way for the past 10+ year is already  good reason to keep it that way. Also what you suggest works mainly for R and C, as soon as you have e.g. a transmission line you need to show at least two parameters (Z<sub>0</sub> and electrical delay) so there needs to be a way to easily understand which is which. And BTW also R and C have additional optional parameters that one may need to see on the schematic .

felix-salfelder, 2019-12-16:
On Mon, Dec 16, 2019 at 01:48:31AM -0800, in3otd wrote:
> https://github.com/Qucs/qucs/tree/develop+qt5-10

there is some more in request #898. it should have gone to
https://github.com/Qucs/qucs/tree/develop+qt5-11. I haven't found the
time yet.

but develop+qt5-10 is obsolete. I will delete it.

> If you want to play with what has been done so far for Qt5 take a look at these branches

any help appreciated, please find build "instructions" in .travis.xml.

thanks
felix

ajohnw, 2019-12-16:
At least on OpenSuse there is no problem compiling providing the qt4lib dev files in installed. The qt4lib is part of the standard distribution but dev files aren't installed by default. There is also a qt3 to qt4 patch library installed.

Not sure what happened with OpenSuse's build. The have an electronics repo. Their build service just checks for dependency errors. I'd guess who ever built it didn't try many features.

StefanBruens, 2019-12-29:
Qt4 was removed from openSUSE Tumbleweed and Leap 15.2 as Qt4 is unmaintained, this removal was announced on the relevant mailing lists. Qt5 is 6 years old already.

Leap 15.1 still has Qt4. The Qt4 and Qt5 packages are completely independent. And of course, runtime libraries and development files (headers, tools, ....) are separate packages (as is the case for almost any distribution).

ajohnw, 2019-12-29:
I haven't installed 15.2 yet and wont for a while. I usually upgrade when the updates stop. This way applications I use are generally available. It would be unusual for OpenSuse KDE to drop support for older programs but that does happen eventually. Their qt5 as mentioned comes with a qt4 library and also one for qt3 specifically for qt5.The 4 maybe the same. Support for GTK apps should also be good but have to admit this has tended to go awol at times over the years. Sad for Linux as being able to use both sources of apps was taken as read once. The only problem I have had in that area is apps that have file indexing dependency. Like a number of KDE users I don't run their or qt file indexing either. Both were active at one point. Maybe there is only one now.

The comments I made on schematic notation where aimed at using the package on larger circuits with lots of components. Nothing more than suggestions based on using schematic capture packages commercially.  Perhaps an answer would be to have an option to not display the R= etc. The lack of justification when items are mirrored about the Y axis is a bug really as the only fix if a problems such as a component  value change over writing the component is to delete and do it again. This messes up component numbering that is provided automatically when copy paste is used.

I've tried several simulation packages now and fact, in many ways Qucs is easily the best. Actually in all ways as far as it goes. ;) I will add a couple of other things to my comment post though. Don't get me wrong I know it's hard and assume that Qucs is mainly aimed at RF and microwave work.

Sadly while I do have lots software experience it's all on automotive ecu's mostly in assembler so can't help. I'm essentially a purely desktop user on a PC google giving me clues if anything else is needed. This is one reason why I use OpenSuse. Software, lib files etc just fire up YAST . If not there use their software search which will also find community versions. I've also used KDE since version 2.something without any problems. Somehow they do manage to keep it stable and always have. Kmail has been embarassing at times but that's about it. Many switched to something else  just as I have. Sylpheed.

closed 2020-02-15 by guitorri

LinArcX, 2023-11-25:
Is there any plan to update to Qt5 or Qt6?
Some distro stop supporting Qucs just because it uses Old qt4.

ra3xdh, 2023-11-25:
@LinArcX look at Qucs-S project https://github.com/ra3xdh/qucs_s/ It supports both Qt5 and Qt6.