Qucs/qucs #942: Op amp simulation problems and one other

ajohnw, 2019-12-16:
I've been trying to use qucs for some audio type work using a tl071 on a single rail. 2 of them in fact. They wont simulate. Go to dual rail and they will. Adding a high pass rc filter to the output also prevents simulation.

One other problem, Net generation failing after changes have been made to the circuit. Error " expected. I think this may have something to do with the way wires behave when they are drawn. Drift off a straight line and it will try and auto correct the path of the wire. I do correct the position of the end of the wire to obtain a straight line but sometimes when I delete a line a couple of small wires are left behind that are hard to select and delete. In fact they only seem to select with a right click. It would be best to disable this attempt at automation and keep the wire straight as the changes it makes generally wont be suitable anyway. Mouse clicks set where the wire changes direction. The net effect seems to be short sections of wire under the one that has been drawn that remain when the one on top is deleted.

Having the net calculation fail is extremely frustrating. The only solution is to throw the lot away and start again. A simple fix might be an option to delete all wires from the schematic and from internal storage so that the user can wire it up again. Big improvement over doing the lot again and this should work whatever has caused the problem.

ajohnw, 2019-12-16:
Argggg. I now do have the simulation running with a single rail so the problem must have been down to something else. I had wondered about the spice convention of using Imeg etc so tried that which cause simulation to fail. Replaced 1meg with 1M and it simulated. I've been using thing like 1k , 1nF from the beginning and also tried adding spaces- 1 k, doesn't seem to matter, So can only put problems down to making circuit changes, deleting wires, adding more and further components etc.

Same with adding a low pass blocking C to the output of the amp. Ok now.

DC analysis is switching to stepping but sadly I didn't save the schematic before making the early changes and may have been so excited that it did simulate that I missed the flashing msg.

I initially wire the amp for single rail and then added a voltage source to provide Vee and later removed it and wired Vee to ground. 

I'm using V0.20

felix-salfelder, 2020-09-26:
> Replaced 1meg with 1M and it simulated.

if 1meg does not work, it should not be allowed. not sure how to fix this at the moment.