Qucs/qucs #943: Sample schematic files 9 simulated added some more parts and 10 didn't

ajohnw, 2019-12-16:
Play 10 produces the expected " error message. Very little editing when I added the changes from 9.
:) Just hope some dev's read these posts. I know it's hard work and thanks for that. Qucs looks to be the best.

[Play9.zip](https://github.com/Qucs/qucs/files/3969237/Play9.zip)
[Play10.zip](https://github.com/Qucs/qucs/files/3969240/Play10.zip)

I was just about to save it under it's intended name.
Version 0.20 on Linux but 0.19 did the same as well.

in3otd, 2019-12-16:
...Qucs is picky.

The error message says `line 63: syntax error, unexpected Identifier, expecting '"'`; open the netlist (press F6), go to line 63, which is
```
C:C6 _net6 _net8 C="47pf" V=""
```
do you see something strange there?
For sure I didn't initially :grin:, then I noticed that the capacitor value is specified as 47p<b>f</b> but it must be written as 47p<b>F</b>.
Similarly C5, 330 n<b>f</b> must be corrected.

It's not necessary to specify the values units but if you do these must be valid units.

ajohnw, 2019-12-16:
Thanks - silly me. I have had the same message before but it could have been down to the same problem. Maybe a silly amp design too but these things have 10^12 input resistance, so may be ok with some high value R's around.

My impression was that it was down to deleting wires and finding little bits with red circles on each end in the form of an L left behind. I feel this is down to running a wire and drifting off straight line and the automatic correction being applied at one end. I corrected one of these when I added extra parts on 10 so wondered. No L was left though this time when the whole wire was deleted.