Qucs/qucs #944: Schematics - oddities rather than bugs + maybe enhancements

ajohnw, 2019-12-20:
I find myself having to often edit a schematic to make space for a component value label and one "feature" in this area. Say I have a vertical resistor symbol with the default long labelling, rotate about the Y axis and later change it to some number such as 5.1k it doesn't rejustify to the component so the value is a long way from it. The only way round this is to change the value before rotating. This may not work as well if say the values was 1k and later changed to 100k as that might overwrite the component symbol.

The component labelling also currently takes up a lot of space. I've been out of electronics for a long time but one popular method used to keep them compact was this sort of thing.
5k1, 5M1, 1R,0R001,5R1. The same thing can be done with inductance and capacitance needing an H and a C. plus the usual m and u, Suppose those could be used on R as well but for some reason schematic packages I have used didn't. The symbol states what type of component it is. Maybe some applications might benefit from a 1uR style as well. The C= etc could be dropped anyway, maybe just don't display them for a quick fix. That would help.. Replacing a decimal point with k etc makes them even smaller also visually clearer. Schematics can be huge compared with the size of a screen.

Supplies to integrated circuits. These could be picked up from supply line labels - directly from the symbol I'd hope. eg an opamp symbol shows Vee and Vcc so adding these labels to the supply would indicate a connection rather than drawing the wires. Usually these would be added to the component as well but they are already there. This could also be used to connect various parts of a circuit. So if the a symbol is added to indicate a connection it could be used for either.

Digital buses - usually these are handled with a single wire and a symbol on each end. Unlikely to be of much interest to me but could well be to some.

The toolbar. I've been treating Qucs like a sketch pad. I make notes. A button on the toolbar rather than having to select paint would be very useful. 

Like many casual users of all sorts of things I generally just use the mouse and don't take the time needed to learn short cut keys other than ones that change the action of mouse clicks. So far I have managed to guess those as needed - cut and paste of circuit blocks.

An oddity in the main dock when components are selected. While there is space for the lot to drop down they don't when when other groups have been used. Say I want to go back to lumped having use another.

I'm impressed. ;) Too many hours spent but would be much quicker now. 2 remaining questions. Can I export in some fashion to import into a pcb package. Some sort of plotter file format might help with that. Pdf's last time I looked were far to verbose. Then how to import various spice components. I used TL071's as the library was there but would also like to compare with TI's own and add more. Some packages like Analogue Devices free one allow this but it seems to be difficult to correct the symbol. Sounds like crap software to me. If one already exists it should be possible to use it and change the name. Pin outs have to be checked manually. That surprised me but it seems that there isn't a standard. I assume the parameters are standard.

ajohnw, 2019-12-21:
I tried to set a log X axis to cover 0 to 10kHz. Doesn't seem to be possible.

I've used qucs model conversion utility and tried some intuitive right clicks to add a model but haven't tried the must create project route.

Docs on this areas suggest creation from a data sheet but the application that seems to be used from Orcad I assume isn't easily available. Might not run on linux. Found that interesting. When I was working in this area Orcad was a choice no engineer wanted to use. They bought something else. My favourite was electronic workbench - too simple to use it seemed.

in3otd, 2019-12-21:
axis with a log scale cannot start from 0, as this starting point should be positioned at -infinity and no screen will be large enough for that :grin: . I guess starting from 1 Hz could be enough, if not specify 1e-3 or lower but something strictly greater than zero.

ajohnw, 2019-12-21:
:) Silly me from 10Hz will do.
Best source of models looks to be LtSpice but suppose they are different yet again. LOL I'm getting irritable. In some if not all ways it's gruesome compared with qucs but will run on Linux.

ajohnw, 2019-12-21:
The only models that are about in quantity looks to be LTspice. There is a community site with more additions as well as what comes with it. Maybe some one can look at the attached jfet model definitions and tell me if they can be converted and what qucs files end with

[standard.jft.zip](https://github.com/Qucs/qucs/files/3991811/standard.jft.zip)

The libraries in qucs are a bit dated. I acually found on of the jfets for sale on the web. Dual in a metal can for £49

ajohnw, 2020-01-02:
Just a few things I noticed.

I saw a request for a cursor probe to look at a dc analysis. I found that even if I zoomed to enormous levels I could read values. I'd be unlikely to want to know all of them and would also be interested in current. Maybe a probe that reads currents on a wire and voltages on a node is possible.

I've tried several OS packages. 2 of these have treated symbols as objects. The model is a characteristic added after the parts has been added - just like resistors and capacitors. This could be extended to multipin packages. The shape may vary such as dil and qfp but pin counts could be a characteristic of a single symbol :( some qfp types are not square.

Then I get disillusioned. I normally do worst case designs. This can only really always be done from data sheets yet no OS package I am aware of allows this to be done - models built from them. There are a few examples of where this may be needed on SystemVisions website apart from the obvious one - no model is available or the one that is uses some odd format or doesn't simulate enough of the various factors. I've already suffered all of these. It doesn't just relate to active devices. Additional data on R's C's an L's is also available from data sheets yet no ability to add them.

Anyway I have enjoyed using Qucs and it's basically a good package but apart from the enjoyment may as well have just breadboarded it and couldn't simulate exactly what I wanted to do due to lack of models so I used 4 op amps.  I wanted to use a few discretes and less op amps. TI's opamp model does not cover enough of the variables and is highly unlikely to use worst case where it does cover them. No more than a toy really. I suspect that the need to be able to generate from data sheets will increase over time. Some are encrypted. Some are application specific. This will probably get worse over time.

felix-salfelder, 2020-09-26:
I agree. May I close this?

closed 2025-03-02 by felix-salfelder