Qucs/qucs #947: Equation-Defined Device cannot handle a^x when x is not an integer >=0

domispace, 2020-01-07:
Hello,
Using Equation-Defined Devices (EDDs) featuring I1=(V1)^x when x is not an integer >=0.
It works for ^0, ^1, ^2, ^n.
It does not work for ^-1, ^-2 or ^(3/4), or ^0.8.
[image]
Does it make sense to modify the EDD capability? i.e, is there another way to produce, for instance, a current that is proportional to the voltage at power(x) which make this request unuseful?
Thanks.

in3otd, 2020-01-07:
I think it does not converge when the exponent is less than 1, where the first derivative has a discontinuity at zero. A quick hack could be to add a small constant so the discontinuity moves elsewhere.
[image]
"Of course" it will likely fail to converge again if you hit the discontinuity.

domispace, 2020-01-07:
Thanks a lot! The trick seems to work.
I have modified the title from "Equation-Defined Device cannot handle a^x when x is not an integer x>=0" to "Equation-Defined Device cannot handle a^x when 0<x<1", because using ^x with x>=1 without the trick (small amount added) works fine.

closed 2020-02-20 by guitorri