Qucs/qucs #950: migrating python2 to python3 test modules

oxmon-2500, 2020-02-01:
qucs-test contains some python modules in python2 syntax. 
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
./contrib/NEWS_crossref.py
./qucs-test/qucstest/__init__.py
./qucs-test/qucstest/colors.py
./qucs-test/qucstest/misc.py
./qucs-test/qucstest/qucsator.py
./qucs-test/qucstest/figures.py
./qucs-test/qucstest/netlist.py
./qucs-test/qucstest/qucsdata.py
./qucs-test/qucstest/qucsgui.py
./qucs-test/qucstest/report.py
./qucs-test/qucstest/schematic.py
./qucs-test/test/test.py
./qucs-test/parse_equations.py
./qucs-test/parse_models.py
./qucs-test/qmat.py
./qucs-test/run_equations.py
./qucs-test/run.py
./qucs/qucs/python/parse_result.py
./qucs/qucs/python/parse_result_example.py
./qucs/qucs/python/qucs-py/qucs.py
Migration already done in local environment

guitorri, 2020-02-03:
Duplicate https://github.com/Qucs/qucs-test/issues/33
Can you make a PR with the changes?

oxmon-2500, 2020-02-04:
OK, after I'm ready with all tests, I will make a PR

oxmon-2500, 2020-02-06:
The changes and tests are done. The changes can be seen as: git diff -w -U0 --color  2c34ef 946204
Somehow I'm not able to make PR. Any suggestion?

guitorri, 2020-02-15:
Fixed on #968

closed 2020-02-15 by guitorri