Qucs/qucs #972: travis strange behavior with Linux cmake Build #3159

oxmon-2500, 2020-02-17:
Travis build terminates with

    .adms.implicit.xml:743: unterminated attribute value - character " expected

The log statements appears double before failure:

    [ 82%] Generating andor4x2.analogfunction.cpp, andor4x2.analogfunction.h, andor4x2.cpp, andor4x2.core.cpp, andor4x2.core.h, andor4x2.defs.h
    [ 82%] Generating andor4x3.analogfunction.cpp, andor4x3.analogfunction.h, andor4x3.cpp, andor4x3.core.cpp, andor4x3.core.h, andor4x3.defs.h
    [info...] admsXml-2.3.6 (e24d007) Feb 17 2020 18:38:04
    [info...] admsXml-2.3.6 (e24d007) Feb 17 2020 18:38:04
    [fatal..] .adms.implicit.xml:743: unterminated attribute value - character " expected

The PR didn't change anything in this area

in3otd, 2020-02-17:
(this is about PR #971)
Rather strange, I've restarted Travis there, let's see...

oxmon-2500, 2020-02-19:
The following run of Travis was OK (without changing anything).

guitorri, 2020-02-20:
Perhaps Travis is building in 2 cores by default? 
Sometimes ADMS can run into race conditions during compilation between some generated files. 
There was a bug like this in the past. Not sure if it was not fixed on the develop branch of ADMS...

closed 2020-02-20 by guitorri

felix-salfelder, 2020-02-20:
On Thu, Feb 20, 2020 at 07:13:11AM -0800, Guilherme Brondani Torri wrote:
> Perhaps Travis is building in 2 cores by default? 
> Sometimes ADMS can run into race conditions during compilation between some generated files. 
> There was a bug like this in the past. Not sure if it was not fixed on the develop branch of ADMS...

ADMS is in Debian for quite some time now [1]. perhaps it is easiest to
just use it in this context, save CPU cycles etc. Not sure what it takes
to make current packages available within travis, should they not be
already.

(the Debian package has been built in non-parallel mode, so it possibly
still has the bug.)

[1] https://packages.debian.org/buster/adms