Qucs/qucs #978: Copyright date in help is still on 2016

Rmano, 2020-03-01:
This is a very minor thing, but... I just compiled the release-0.0.20 branch on Ubuntu 18.04 (without problems --- great! Thanks everybody!) and on "About..." I have:

[image]

...I think it's worth correcting, but I don't think it's worth a PR... ;-)

in3otd, 2020-03-01:
oops :grin:
I think there might be other places with the wrong year (I think I saw something in the CMakefiles @guitorri recently modified).

And by the way, I'd maybe put Stefan Jahn name too in the second (C) line, as he wrote the simulator (`qucsator`), which is still the only simulator Qucs supports...

felix-salfelder, 2020-09-27:
>  ...I think it's worth correcting, but I don't think it's worth a PR... ;-)

done.

yes, there might be other places, and yes Stefan Jahn wrote qucsator. open new tickets please.

closed 2020-09-27 by felix-salfelder