Qucs/qucs #979: transistor model problems

boomboompsh, 2020-03-01:
the BFR92A is an NPN transistor, not PNP. 
datasheet: [BFR92A](https://www.nxp.com/docs/en/data-sheet/BFR92A_N.pdf)
I don't understand the transistor models yet, so it could be as easy as switching it from "pnp" to "npn". The X5_500 example uses this part, so it needs to be changed to use the BFT92A, the pnp equivalent.

The MMBTH10 and MMBTH24 are not characterized above 1GHz on their respective datasheets, but the component library claims frequencies of 650GHz and 400GHz respectively. It's probably a typo and it was supposed to be MHz. I'm not sure exactly what the frequencies listed on the model description are supposed to mean.
[transistor library](https://github.com/Qucs/qucs/blob/0ef64c307d72fa9c3d8075d6a69b79146f060e40/qucs/qucs-lib/library/Transistors.lib)
[X5_500 schematic](https://github.com/Qucs/qucs/blob/0ef64c307d72fa9c3d8075d6a69b79146f060e40/qucs/examples/X5_500.sch)

boomboompsh, 2020-03-29:
The BUZ272 has an on resistance of 0.3 Ohm, not 0.3 mOhm as claimed in the component library.
[BUZ272 datasheet](http://pdf.datasheetcatalog.com/datasheet/infineon/1-buz272.pdf)

oxmon-2500, 2020-03-30:
Thank you for contribution, PR #996 issued

felix-salfelder, 2020-09-26:
closing, as this is covered by 996.

closed 2020-09-26 by felix-salfelder