Qucs/qucs #980: "linblocks" tutorial is not in the menu...

Rmano, 2020-03-01:
Hi, this is probably my fault, but I do not know how to correct it. I have seen that my contributed tutorial about linear blocks and frequency responses has been added (thanks!), but it still doesn't appear in the menu: 

[image]

Probably I was supposed to add it somewhere in the sources, but I do not know where... 

Thanks!

guitorri, 2020-03-01:
Is the file being installed? The menu is filled with whatever files are on the DocDir.

Rmano, 2020-03-01:
Hmmm... I don't know. I will try to check - should I had put something special in the Makefile or anywhere?

Thanks for helping - - it's not really critical so don't waste time on it, I'll try to look at it when I get some time (probably on Tuesday)

guitorri, 2020-03-01:
I checked the Makefile.am, it seems fine...

felix-salfelder, 2020-09-26:
> it's not really critical so don't waste time on it

defer to 0.0.21?

Rmano, 2020-09-26:
It's ok for me. 
Even better if you change the release to, say, 0.9.0 (`/me ducks`, ;-) --- I know I have already said that, but it's my pet peeve...)

felix-salfelder, 2020-09-26:
Let's discuss version numbers once 0.0.20 is out. Fingers crossed it will be relevant.

closed 2020-09-26 by felix-salfelder