Qucs/qucs #983: Problem with equation

A2va, 2020-03-04:
By clicking on Add this does not add a line to the equation while in some examples several lines are displayed
[image]

[image]
Did i do something wrong or is it a bug ?

in3otd, 2020-03-04:
did you first type an equation with a different independent variable (i.e. not `y`, which is already defined above) and then press *Add*?

A2va, 2020-03-05:
Ok I understand now how it works I didn't know to remove the 'y' and press Add.

oxmon-2500, 2020-03-05:
I played around and there is a simple solution to make this dialog more user friendly: 
Add button adds an empty row and clears the input fields. After filling the input fields Add or Enter fills the empty row. 
I make a branch with it. If someone review/test it, I will make a PR
https://github.com/oxmon-2500/qucs.git  branch **qucs_983_editProperties**
Affected files: componentdialog.cpp, componentdialog.h

felix-salfelder, 2020-09-26:
> If someone review/test it, I will make a PR

I have looked at qucs_983_editProperties. Please create a merge request.