Qucs/qucs #986: Transient simulation with switches

danyill, 2020-03-07:
I guess this is a support/diagnostic request.

I like Qucs very much and mainly use it for transient simulations.

The attached three-phase model of a power system capacitor bank is where I am simulating a shorting of a capacitor bank layer followed by a fuse operation (opening a capacitor element). Currently I have the switch occurring at 8 seconds which is where it hangs. Very occasionally I can fiddle the parameters to have the simulation run, however often it doesn't.

I go through an extensive process of trying to make it work which is mostly superstition:

* The 0.1 ohm resistors are to make it look "real world" and deal with potential "charge" sloshing at very high current values during transient events (which may not help convergence).
* R12-R19 (1e12 ohms) are parasitic elements to help "convergence".
* The main problem is the shorting and fuse operation (S4 and S3) tend to mean the transient simulation just hangs. Occasionally by changing the switch resistance when on and the timestep I manage to get a combination which works.

I've spent several days learning how sensitive the transient simulation is to small changes and I'm getting to a stage where without a process to "help" convergence I may need to give up.

I'd appreciate any help or advice on steps to try when I have these problems to try and get the simulation going again. I don't think there is much in the way of qucsator debug out put at the moment as far as I'm aware.

I've attached the circuit diagram (renamed form `sch` to `txt`):

[HAM_C1_Unbalance_COMTRADE_test16.txt](https://github.com/Qucs/qucs/files/4301289/HAM_C1_Unbalance_COMTRADE_test16.txt)

Ideally I'd also like to reduce the timestep to between 500 us and 125 us but I cannot get the transient simulation to run under these conditions.

danyill, 2020-03-07:
There is a very strange phenomena at play. Transient simulation convergence appears to depend on whether the word "Ohm" is provided on resistors.

An example:

* [HAM_C1_Unbalance_COMTRADE_test19_succeeds.txt](https://github.com/Qucs/qucs/files/4301346/HAM_C1_Unbalance_COMTRADE_test19_succeeds.txt)

* [HAM_C1_Unbalance_COMTRADE_test19_fails.txt](https://github.com/Qucs/qucs/files/4301345/HAM_C1_Unbalance_COMTRADE_test19_fails.txt)

These should be identical except in the first one I have removed the word "Ohm" from resistors.

oxmon-2500, 2020-03-07:
Strange, Your schematics begin win <Qucs Schematic **0.0.22**> You are ahead 2 versions of the qucs ;-)

felix-salfelder, 2020-03-07:
On Fri, Mar 06, 2020 at 06:23:45PM -0800, danyill wrote:
> I've spent several days learning how sensitive the transient
> simulation is to small changes and I'm getting to a stage where
> without a process to "help" convergence I may need to give up.

Thanks for using free software for circuit simulation.

> I'd appreciate any help or advice on steps to try when I have these
> problems to try and get the simulation going again.

This is more difficult. In a nutshell, qucsator does not implement this
correctly. you could try gnucsator, a qucs frontend for gnucap, a mixed
mode simulator. it does time step control and discrete events required
for transient simulation with switches.

> I've attached the circuit diagram (renamed form `sch` to `txt`):

Please share the flattened circuit (in "netlist.txt").

> Ideally I'd also like to reduce the timestep to between 500 us and 125
> us but I cannot get the transient simulation to run under these
> conditions.

there is a tradeoff between speed and accuracy. in gnucap you choose the
accuracy, and the timesteps are automatic.

danyill, 2020-03-07:
True, I was seeing if Qucs-S would do any better, but alas no.

in3otd, 2020-03-07:
um, strange; I was about to suggest to use Qucs-S with the `ngspice` simulator. I'd expect `ngspice` to be able to simulate that circuit without any issue.

@danyill Both the examples you provided work fine here; this is unfortunately quite typical for difficult simulations as these - just small numerical errors due to compiler differences can make a difference here.
FYI, in Qucs the more robust simulations are S-parameters and AC, transient has some issues with time step control, as you saw. In the long term we would like to have the Qucs GUI to work as a frontend to other simulators too (`ngpsice`, `gnucap`, etc.) but there are not enough active developers to achieve this at present.

@felix can your provide more info about how to use `gnucsator` with the Qucs GUI?

felix-salfelder, 2020-03-07:
On Sat, Mar 07, 2020 at 12:10:36AM -0800, in3otd wrote:
> @felix can your provide more info about how to use `gnucsator` with the Qucs GUI?

From the perspective of Qucs (GUI), the gnucsator simply replaces qucsator
(The binary executable). when invoked like

$ QUCSATOR=gnucsator.sh qucs

Qucs will run the other (no other changes needed or intended) c.f. [1].
It will make some things work better (e.g. transients and additional
components), but others are missing.

But there is not a single way. You could as well use Qucs to build a
netlist.txt (I did not implement .sch parsing on this end). pick it up
from there with gnucap-python, and draw the plots with matplotlib or so.

[1] https://github.com/Qucs/gnucsator/blob/master/README

danyill, 2020-03-07:
Thank you @felix-salfelder @in3otd 

@in3otd 

Thank you.

* I have been switching between two Linux machines and two Windows machines and wondering whether I am going slightly crazy when what works on one does not work on the other!

* I'm not sure if ngspice allows the two switches I have in the model -- it removes them when I change the simulator engine and I don't know much about spice and could not figure how to put them back.

@felix-salfelder 

Thank you.

* The netlist is here: https://gist.github.com/danyill/3648994d3aadf060ee910c84ced080ed. Sometime I have better luck when the switch is set to `abrupt` instead of `spline` but not in this case.

* I may be doing something silly. I try to alternate the switch time versus the simulation step from time to time which sometimes helps (as long as it is not too small).

* I did try qucsator.sh after sorting out some compilation issues. Unfortunately this hangs for me and I'm not sure how to progress -- a paste from the simulation screen is here: https://gist.github.com/danyill/5ed5b72607f197e09c5fb8b24c7695d0

oxmon-2500, 2020-03-07:
@danyill  Thanks for information, I didn't know there is such spin-off

guitorri, 2020-03-07:
Thank you for the reproducers.

On my box `test16` is trowing an assertion:
```qucsator: /home/guilherme/git/qucs/qucs-core/src/components/tswitch.cpp:196: virtual void tswitch::calcTR(double): Assertion `r >= ron' failed.```

Both `test19_succeeds` and `test19_fail` are running till the end.
I am running latest `develop` on Ubuntu 18.04.

There are several issues we don't fully understand, like
- platform dependent issues, like #205 
- switching circuits, like #41

felix-salfelder, 2020-03-07:
On Sat, Mar 07, 2020 at 01:33:08AM -0800, danyill wrote:
> * The netlist is here:
> https://gist.github.com/danyill/3648994d3aadf060ee910c84ced080ed.
> Sometime I have better luck when the switch is set to `abrupt` instead
> of `spline` but not in this case.

Thanks. There are parameters such as time="[1000 ms;5ms;20000 ms]" in
the netlist. parameters are numerical in gnucap (and in verilog), so
this hack does not work. the correct way to model this is either
multiple switches (each one only switching once), or with a switch
controlled by a variable continuous source (e.g. a sine wave).

The switch for qucs is implemented as a spice macro with a voltage
source and a controlled switch. see [1].

> * I did try qucsator.sh after sorting out some compilation issues.
> Unfortunately this hangs for me and I'm not sure how to progress -- a
> paste from the simulation screen is here:
> https://gist.github.com/danyill/5ed5b72607f197e09c5fb8b24c7695d0

When I run
$ gnucsator.sh -i netlist.txt
I get 34524 transient steps in about 1 second.

The transient output is stored in netlist.txt.tmp.tr. It looks right to
me modulo the switchng model issue above, but it is not in a format that
Qucs understands. the conversion currently is a slow shell function in
gnusator.sh, and will be replaced (WIP). If you want to bypass the
conversion, comment out the line

[ -f $out.tr ] && cat $out.tr | postproc >> $outfile

in gnucsator.sh.

[1]
https://git.savannah.gnu.org/cgit/gnucap/gnucap-plugins.git/tree/include/nonlinear.v?h=qucs&id=2fa836fba1c60f21bcaa28ba774a5d7939dd785e

danyill, 2020-03-08:
@felix-salfelder thank you for this advice. I think I understand.

I believe this bash file does not work as intended for the post processing of the data (it continued for 45 minutes without completing in my simulation) but it does seem as though `gnucastor` gives better results and I may be able to use it together with Qucs from the looks of it (it would be lovely if there was CI, a tagged release and it was pre-built for different platforms because I'll need to build for Windows which may not be easy).

To make this work from Qucs in an easy to use fashion I put together a simple Python script which processes the data into qucs and csv output. In case it's of any use to someone else I include it below:

```python
#!/usr/bin/env python3

import sys
from pathlib import Path

import pandas as pd

SPACE = '  '

input_file = Path(sys.argv[1])
output_file = Path(sys.argv[2])

df = pd.read_csv(input_file, sep='\s+', header=0)

print('Reading from file: ' + str(input_file))

output_file_csv = output_file
df.to_csv(output_file.parent / Path(output_file.stem + '.csv'))

dep_variables = [v for v in list(df.columns.values) if v != '#Time']

print('Variables: ' + ",".join(dep_variables))

def write_values(dataset, type='indep', name='time', datasetname='#Time'):
  output = [ ]

  if name != 'time':
    output.append('<' + type + ' ' + name + ' ' + 'time' + '>')
  else:
    output.append('<' + type + ' ' + name + ' ' + str(len(df[datasetname]) - 1) + '>')
  
  for r in dataset[datasetname].iteritems():
    output.append(SPACE + str(r[1]))

  output.append('</' + type + '>')
  output.append('')
  return "\n".join(output)

print('Writing to: ' + str(output_file))

with open(output_file, 'w') as f:
  f.write('<Qucs Dataset >')
  f.write('\n')
  f.write(write_values(df, 'indep', 'time', '#Time'))
  for v in dep_variables:
    f.write(write_values(df, 'dep', v, v))
```

and then I replace the line in `gnucsator.sh`

```bash
[ -f $out.tr ] && cat $out.tr | postproc >> $outfile
```

with a reference to the python script (which is on the pa

```bash
[ -f $out.tr ] && cat $out.tr | qucs_data_convert.py $out.tr $outfile
```

Thanks everyone for helping out. I'm looking forward to when transient simulations work nicely. 

Perhaps for future it would be worth detecting some of the problematic conditions and giving a warning to the user or suggest using gnucsator. When the tool cannot provide a warning and sometimes works it is challenging to non-technical end users (like me, who keeps thinking I'm doing something wrong).

I'm happy for this issue to be closed if that's house rules here.

closed 2020-03-08 by felix-salfelder