Qucs/qucs #1002: No Zener Diode available in Components Database

derei, 2020-04-21:
Yeah, as title says: couldn't find Zener Diode anywhere. Would be useful to cover all essential electrical components.
Thanks.

andresmmera, 2020-04-22:
For commercial parts, please go to the component library (CTRL+5) and select "Z-diodes" in the combobox.

The model of the Zener diodes is the same as for conventional diodes, but some parameters (specifically BV, Ibv) vary.

BV is the reverse voltage breakdown and Ibv the reverse current.

I hope this helps

oxmon-2500, 2020-04-22:
[image]

oxmon-2500, 2020-04-22:
@derei  Be carefull unsing the following Z-Diodes:

    Z-Diodes 1N5342B 'Vj' occurred 2x
    Z-Diodes 1N5344B 'Vj' occurred 2x
    Z-Diodes 1N5346B 'Vj' occurred 2x
    Z-Diodes 1N5377B .Def:Z-Diodes_1N5377B -> .Def:Z_Diodes_1N5377B

There are typos in the definitions. Using it in a circuit causes errors. Will be solved by Pull Request #993

oxmon-2500, 2020-04-22:
Since the lack of Z-Diode in nonlinear component may be confusing, an idea for an improvement: 
Z-Diode symbol will be included in the nonlinear components, but clicking on it opens a library selection

derei, 2020-04-22:
thank you. i was using the search function for "diode" and and "zener", never occurred to me to search for z-diode.

felix-salfelder, 2020-07-17:
there is an idea for an improvement, but the original issue has been solved.

closed 2020-07-17 by felix-salfelder