Qucs/qucs #1003: Can't run simulation

derei, 2020-04-22:
I tried following some tutorials on YouTube (unfortunately, all of them use outdated qucs versions), but for some reasons, I can't get the same results, my simulation won't result in any meaningful values in the table. 

I'd appreciate any help right now (if you know about some tutorial using a newer version of qucs).

oxmon-2500, 2020-04-22:
Pls attach the schematic file (name.sch) of your try

derei, 2020-04-22:
[image]
sch file attached here:
[test.zip](https://github.com/Qucs/qucs/files/4517406/test.zip)

oxmon-2500, 2020-04-22:
After every change the values, simulate button must be pressed again,  dpl window closed. Simulation do not start automatically.

derei, 2020-04-22:
Thanks, anyway it looks like dpl tab doesn't have to be manually closed. Pressing Simulation would update the values.

I've got this warning: WARNING: DC1: inserted virtual resistance at node `_net1' connected to [R1,Voltage,V1]

Not sure what it means

oxmon-2500, 2020-04-22:
The circuit must be connected to ground
[image]

derei, 2020-04-22:
Any proper documentation available to explain this software? Help seems basic.
For example, what is the purpose for the dpl tab that gets created with the simulation? the whole window looks empty, nothing to show...yet every time when simulation button gets pressed, the software switches to it

miguel9554, 2020-05-02:
@derei, I'm new to Qucs as well as you. I found the [official documentation](http://qucs.sourceforge.net/docs.html) to have useful resources. The [Getting Started](http://qucs.sourceforge.net/docs/tutorial/getstarted.pdf) tutorial offers a nice introduction to the sotware and how to use it. You alse have specific tutorials for each topic.

closed 2020-09-26 by felix-salfelder