Qucs/qucs #1006: develop branch is NOT Qt5 compatible?

jiapei100, 2020-05-07:
develop branch is NOT Qt5 compatible?

thmalmeida, 2020-06-02:
ubuntu 20.04 has dropped Qt4. Now, to keep Qucs working I could add Qt4 lib from PPA repository. Available on https://launchpad.net/~rock-core/+archive/ubuntu/qt4.

Rmano, 2020-06-12:
I think that lot of people will move to 20.04 or newer distribution, and QT4 will be dropped and will be more and more difficult to find. As a stopgap, I think that trying to build a snap, or AppImage version would be nice. I will look into it as I have time, but it's quite difficult now --- I [did a snap package for QUCS when the same problem appeared on 16.04->18.04](https://github.com/Rmano/qucs-19-snap), but it wasn't easy and it wasn't really working so well (PDF export was broken, for example).

Is there anybody here who is expert on snap, appimages or similar kind of packages? Please contact me.

Switching to QT5 is not easy, it seems...

Rmano, 2020-06-12:
> ubuntu 20.04 has dropped Qt4. Now, to keep Qucs working I could add Qt4 lib from PPA repository. Available on https://launchpad.net/~rock-core/+archive/ubuntu/qt4.

...is that repo sufficient for building QUCS? The problem will be to keep up compiling new versions...

Rmano, 2020-06-14:
I have managed to create an AppImage for QUCS-0.0.20-pre2. You need the patch in PR #1011 

These are the steps I followed:

1. get the source from https://github.com/Qucs/qucs/releases/tag/qucs-0.0.20-rc2

2. now I created a directory for it (sorry it's a bit long) and unpacked and built it:
```
cd /home/romano/software/qucs/AppImage/
tar xvzf ../qucs-qucs-0.0.20-rc2.tar.gz
cd qucs-qucs*
mkdir build
cd build 
cmake .. -DCMAKE_INSTALL_PREFIX=/usr
```

3. After the compilation went ok, I prepared the `AppDir`:
```
make install DESTDIR=AppDir
```

4. get the deploy application:
```
wget https://github.com/linuxdeploy/linuxdeploy/releases/download/continuous/linuxdeploy-x86_64.AppImage
chmod +x ./linuxdeploy-x86_64.AppImage
```
5. ...and deploy it. The most tricky thing is the `LD_LIBRARY_PATH` thing

```
export LD_LIBRARY_PATH=/home/romano/software/qucs/AppImage/qucs-qucs-0.0.20-rc2/build/AppDir/usr/lib
./linuxdeploy-x86_64.AppImage --appdir AppDir --output appimage
mv ./Qucs-x86_64.AppImage  ./Qucs-0.0.20-pre2-x86_64.AppImage
```

After that, I moved the AppImge to a VM with Ubuntu 0.0.20, install `adms` from the standard repo, and...

[image]

Lightly tested, seems to work.

Now, it will be great is somebody more knowledgable than me on the Travis or whatever CI is able to generate this kind of thing automatically on releases. It will be a life-safer waiting for the QT5 version.

References:
https://docs.appimage.org/packaging-guide/from-source/native-binaries.html#packaging-from-source
https://github.com/linuxdeploy/linuxdeploy/issues/76

Rmano, 2020-06-14:
If you like to live dangerously, I have a copy of it in: https://www.rgtti.com/blobs/Qucs-x86_64-0.0.20-pre2.AppImage

probonopd, 2020-06-27:
@Rmano thanks a ton, seems to run well on Xubuntu 18.04.4 LTS.

Rmano, 2020-06-27:
> @Rmano thanks a ton, seems to run well on Xubuntu 18.04.4 LTS.

Well, in 18.04.4 you can compile it natively, and it will run better... the point is to check it 20.04 or in (most) modern distribution that have dropped QT4.

tinywrkb, 2020-07-03:
Initial Flatpak packaging can be found [here](https://github.com/tinywrkb/flatpaks/blob/master/com.github.qucs.qucs/com.github.qucs.qucs.yaml).  
It needs [Qucs base app](https://github.com/tinywrkb/flatpaks/blob/master/com.github.qucs.qucs.BaseApp/com.github.qucs.qucs.BaseApp.yaml) for building so build and install the base app first. The reason for having a base app is to be able to incorporate Qucs in a Qucs-S package without needing to rebuild it.

Rmano, 2020-07-12:
Ok, now I have a AppImage developed on 16.04, that should work on a lot of systems. If you can check it, I will be grateful.

https://rmano.github.io/qucsAppImagesBuild/

Rmano, 2020-07-12:
BTW, I am thinking about submitting it here: https://github.com/AppImage/appimage.github.io#how-to-submit-appimages-to-the-catalog  --- but I'd like a thumb-on from the developers and, if possible, a bit of feedback about the working status on several distros. 

I think it could be a nice stop-gap until the Qt5 migration is finished. I will try to maintain it live, i.e., if there is a new release (-rc3 or whatever) I will update the repo.

cederom, 2021-02-08:
`qucs` port was deleted from FreeBSD in March 2019 because it is not Qt5 compatible and Qt4 was removed from the OS. This utility looks amazing, would be really nice to have it working with Qt5 or Qt6 :-)
https://www.freshports.org/cad/qucs/

thomaslepoix, 2021-02-08:
I anwsered you on #560, please do not repost the same question on multiple issues.

Nikita-Presnov, 2021-12-16:
> Switching to QT5 is not easy, it seems...
Hm...
Most of the time I just edited CmakeList.txt a bit when I needed a qt5 port. Your program looks complicated, but I don’t think it’s such a daunting task. Moving to qt6 promises to be even easier.

Rmano, 2021-12-20:
@Nikita-Presnov  it would be great if you could help with it. I think that the change is not trivial at all (there are still things from qt3 in the codebase, I think). I use QUCS a lot but I am not a developer, so I am stuck with the AppImage I made. I am sure that the developer will appreciate it!

Nikita-Presnov, 2021-12-20:
Sorry, i forgot about it.

felix-salfelder, 2024-07-02:
qt5 port in develop now.

closed 2024-07-02 by felix-salfelder