Qucs/qucs #1008: qucsconv not converting multiple subckt spice file

oxmon-2500, 2020-05-31:
origin: [see](https://sourceforge.net/p/qucs/mailman/message/37025370/)

Hello,

I am trying to use a TI Current Sense Amp in a qucs circuit. I have
followed the steps to convert the spice files to the qucs circuit file type
however it errors out.

I then attempted to run the command from the command line with:

qucsconv -i INA191_TINA_AIO/INA191A5.lib -o INA191A5.out -if spice -of qucs

And I am met with the error:

*line 224: syntax error, unexpected Identifier, expecting ENDS_Action*

This happens for all TI Spice files I have attempted to import. I have
verified the Spice files can work with ngspice, they are loaded and
simulated without issue in ngspice. I have also tried a simple single
subckt spice file as a sanity check. This can be converted with qucsconv.

I would appreciate the help. I am using the latest master branch as of
05/30/20
690d2ad9a859c50d79ea9d56ad52b4e370ea9c54

The TI INA191 spice files can be downloaded here:
https://www.ti.com/lit/zip/slym110

I appreciate any and all help.

Best,
Andrew Smith
*M.S. Computer Engineering*
*Passat Research Group*
*atsmith3@... <atsmith3@...>*
-------------------------------------------------------------------------------------------------------------------
It seems that the .PARAM name = value is not known to qucsconv (syntax def. in parse_spice.ypp, parse_spice.lpp) . Also constants like {PWR(0.1u,0.5)/1E11}. I suppose the syntax of spice in qucsconv is outdated.

Workaround: 
- Comment out the .PARAM line and substitute all occurances of name with the value.
- Replace constants like  {PWR(0.1u,0.5)/1E11} with the actual value
Its imaginable to do a script doing it automatically

atsmith3, 2020-05-31:
Great, thanks! I will do the workaround for now.

felix-salfelder, 2025-01-18:
"Spice syntax" is an open problem everywhere. Lacks essential features, and is still in use. A moving target.

Possible solution is a converter that reads Spice .subckt declarations (given a dialect) and emits Verilog. Maybe not qucsconv?