Qucs/qucs #1009: Qucs crashing as soon as any component is added.

rolimans, 2020-06-07:
As soon as I add any component to a sketch file (eg: a resistor) the program crashes.
The log of the execution follows:

```
Unable to load library icui18n "Cannot load library icui18n: (icui18n: cannot open shared object file: No such file or directory)" 
Warning: QVariant::save: unable to save type 261.
[1]    13495 segmentation fault (core dumped)  qucs
```
I'm using qucs 0.0.20 builded from source in Arch Linux 64 bits.
PS: icui and icui32bit are both installed.

oxmon-2500, 2020-06-09:
try following:

    sudo find / -name "libicui18n*"
    strace -e trace=all qucs &>qucs_strace.log
    grep libicui18n qucs_strace.log

and post the output of grep.

perhaps 

    export LD_LIBRARY_PATH=path_to_libicu18n 

will help

rolimans, 2020-06-09:
> try following:
> 
> ```
> sudo find / -name "libicui18n*"
> strace -e trace=all qucs &>qucs_strace.log
> grep libicui18n qucs_strace.log
> ```
> 
> and post the output of grep.
> 
> perhaps
> 
> ```
> export LD_LIBRARY_PATH=path_to_libicu18n 
> ```
> 
> will help

Strace log is:

```
openat(AT_FDCWD, "/usr/lib/tls/haswell/x86_64/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/haswell/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/x86_64/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/tls/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/haswell/x86_64/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/haswell/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/x86_64/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libicui18n.so.64", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/libicui18n", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
```

My libraries are at:

```
/usr/lib/libicui18n.so
/usr/lib/libicui18n.so.67.1
/usr/lib/libicui18n.so.67

/usr/lib32/libicui18n.so
/usr/lib32/libicui18n.so.67.1
/usr/lib32/libicui18n.so.67

```

It seems qucs is looking for the wrong library extension (.so.64). Why doesn't it just look for .so?

felix-salfelder, 2020-06-09:
On Tue, Jun 09, 2020 at 12:29:58PM -0700, Eduardo Amaral wrote:
> My libraries are at:
> 
> ```
> /usr/lib/libicui18n.so
> /usr/lib/libicui18n.so.67.1
> /usr/lib/libicui18n.so.67
> 
> /usr/lib32/libicui18n.so
> /usr/lib32/libicui18n.so.67.1
> /usr/lib32/libicui18n.so.67
> 
> ```
> 
> It seems qucs is looking for the wrong library extension (.so.64). Why doesn't it just look for .so?

Note how *.so is only used during linking. afterwards your program uses
the library that you have linked against (c.f. SONAME). the general
approach is, not to delete a (shared) library after linking against it.

Now, I am not exactly sure why qucs starts at all, given that a library
is not there. maybe something else is loaded dynamically, and is trying
to use the library, which has disappeared unexpectedly. broken Qt
installation? (need more context).

rolimans, 2020-06-09:
It's not me the only one to have this problem in Arch Linux. In fact I was advised to create this issue in the AUR forum. I also don't understend why the program launches at all. And I have no idea if the crashes are related to the missing library. Still have not figred out how to make qucs find libcui18n

oxmon-2500, 2020-06-10:
On my centos8 system it looks like:

    /usr/local/Trolltech/Qt-4.8.7/lib/libicui18n.so.60 ... No such file or directory
    /lib64/libicui18n.so.60", O_RDONLY|O_CLOEXEC) = 3
    /usr/local/Trolltech/Qt-4.8.7/lib/libicuuc.so.60 ... No such file or directory
    /lib64/libicuuc.so.60", O_RDONLY|O_CLOEXEC) = 3
    /usr/local/Trolltech/Qt-4.8.7/lib/libicudata.so.60 ... No such file or directory
    /lib64/libicudata.so.60", O_RDONLY|O_CLOEXEC) = 3

I think its a problem of Qt, which tries to loads the specific library dynamically.

The hack (not tested)

    mkdir ~/libhack
    ln -s /usr/lib/libicui18n.so.67.1 ~/libhack/libicui18n.so.64
    ln -s /usr/lib/libicuuc.so.67.1 ~/libhack/libicuuc.so.64
    ln -s /usr/lib/libicudata.so.67.1 ~/libhack/libicudata.so.64
    export LD_LIBRARY_PATH=~/libhack:$LD_LIBRARY_PATH
    qucs

will probably do the trick, presumed the version 67 is downwards compatible with version 64

Dantali0n, 2020-07-16:
I'm having the exact same issues on Manjaro (Arch).

Rmano, 2020-07-16:
It's probably more of a workaround, but you can try to see if the AppImage from https://rmano.github.io/qucsAppImagesBuild/ works. It's minimal (no documentation nor tests nor examples) and it has been build on an Ubuntu 16.04.6 system (where Qt4 is not legacy...).

AlexanderKnapik, 2020-09-11:
I've run into this issue many times before all using Manjaro Linux. I'll use the AppImage for now as a workaround.

felix-salfelder, 2020-09-26:
It seems this has been traced down to qt incompatibilities and runtime issues. Lacking a solution, we still have a workaround -- thanks to Rmano.

closed 2020-09-26 by felix-salfelder