Qucs/qucs #1012: Version String Missing in .dat Files in Qucs Version 0.0.22

cweickhmann, 2020-06-17:
I observed that in Qucs 0.0.22, the `<Qucs Dataset >` tag is not complete. In particular, it is missing the version string (in this case 0.0.22) which makes e.g. `qucsconv` refuse working.

**MWE**
* Create file (here `single_coil.sch`)
* Place any simple setup in there (in my case: DC source, resistor, GND, connect them, add a DC Simulation)
* Check the `single_coil.dat` file.

In 0.0.19, the first line reads
```
<Qucs Dataset 0.0.19>
```

In 0.0.22, the first line reads
```
<Qucs Dataset >
```

This makes qucsconv hickup and refuse to work in a on first sight rather cryptic error message (`line 1: syntax error, unexpected '<', expecting Version`).

Rmano, 2020-06-18:
Hmmm... QUCS is at 0.0.20. Are you mixing this project with https://github.com/ra3xdh/qucs_s/tree/0.0.22 ?

cweickhmann, 2020-06-18:
Yup. You're right.

closed 2020-06-18 by cweickhmann