Qucs/qucs #1016: BC547 collector overcurrent

dealmeal, 2020-10-14:
Simulating a switch with a BC547 in QUCS 0.0.19 I have noticed that the collector current reaches 265 mA. However, the maximum collector for such transistor is 100 mA according almost manufacturers datasheet and Horowitz bible. Was not expected that simulator crash and raise some error mesage?

Rmano, 2020-10-15:
No, I don't think it's expected unless you do a power limit simulation which isn't in the simulation sets of QUCS. I think I never saw this, even in SPICE. 
It is similar to the small-signal analysis for the frequency response --- you will have a linear answer no matter what the amplitude is, even if in real life the thing will saturate --- the analysis is done with the linear equivalent!

[image]

(Well, QUCS is doing something different from SPICE here, because if you push it too much it "switches" to a strange result...)

But basically, you have to take into account the model you are using; the simulation will use that model that invariably has differences with the real-life component. I am not aware of models that take into account maximum ratings, not in open-source simulators at least.

felix-salfelder, 2020-10-15:
On Wed, Oct 14, 2020 at 02:04:06PM -0700, dealmeal wrote:
> Simulating a switch with a BC547 in QUCS 0.0.19 I have noticed that
> the collector current reaches 265 mA. However, the maximum collector
> for such transistor is 100 mA according almost manufacturers datasheet
> and Horowitz bible. Was not expected that simulator crash and raise
> some error mesage?

It's either a qucsator bug or a model limitation, both we should keep
track of.

Could you please share a minimal example netlist?

Rmano, 2020-10-15:
> It's either a qucsator bug or a model limitation, both we should keep track of.

Well, probably a model limitation, but I don't know how to approach it. Stopping the simulation? I do not think any simulator I worked with *ever* did that. Maybe a warning in the log (as we have now for un.physical values of parameters, try putting η=4 in a diode ;-)). Again, it's true that I stopped used commercial simulators 20+ years ago, and I never saw it... the reference SPICE model for BC457 seems this one:

```
*PVDB
*DATABOOK PHILIPS
.MODEL BC547B  NPN (BF=530 NE=1.3 ISE=9.72F IKF=80M IS=20F VAF=50V
+      BR=10 NC=2 ISC=47P IKR=12M VAR=10
+      RB=280 RE=1 RC=40 TR=.3U
+      CJE=12P VJE=.48 MJE=0.5 CJC=6P VJC=.7 MJC=.33 TF=.5N)
```

no maximum current anywhere here.

Rmano, 2020-10-15:
...or maybe a "warn over" parameter could be added to current, voltage and power probes. That would be really useful in a lot of cases, and a new thing for simulators, at that.

felix-salfelder, 2020-10-15:
On Thu, Oct 15, 2020 at 02:33:23AM -0700, Romano Giannetti wrote:
> ago, and I never saw it... the reference SPICE model for BC457 seems
> this one:
>
> ```
> *PVDB
> *DATABOOK PHILIPS
> .MODEL BC547B  NPN (BF=530 NE=1.3 ISE=9.72F IKF=80M IS=20F VAF=50V
> +      BR=10 NC=2 [...]

thanks. have you got something that runs in qucsator (i.e. a netlist.txt)?

This "SPICE model" is not a model but just a parameter set. The model
referred to here is "NPN", and it could be anything. This is relevant as
behaviour and defaults seem to vary across versions, implementations and
simulators.

> no maximum current anywhere here.

If other simulators limit the current, then probably because it is
implicit. It all depends on the model. In some SPICE implementation you
could specify a "model order" to determine what "NPN" means, I don't
think qucsator has this. Sure, I can think of another simulator that
lets you pick a model.

Rmano, 2020-10-15:
> If other simulators limit the current, then probably because it is implicit.

I don't think so. As I said, I *never* saw a simulator with this option. It would be awkward anyway; the 100 mA maximum is not a device hard limit, each component will behave a little different, and most will work perfectly over that linit. I think this is outside the task of a simulator... and in the hand of the designer (I need to be paid for something, no ;-P?). Although a warning coud be nice.

The SPICE model order is used especially in MOSFET, as a compromise between correctness and speed. I do not know the innards of qucsator, but there is somewhere a doc describing the model for BJT, I remind reading it. The SPICE one is here for example: https://class.ece.uw.edu/cadta/hspice/chapter_14.pdf (this one is the HSPICE model, which is a better implementation especially for high-power transistors, but well...)

felix-salfelder, 2020-10-15:
On Thu, Oct 15, 2020 at 03:35:28AM -0700, Romano Giannetti wrote:
> > If other simulators limit the current, then probably because it is implicit.

> I don't think so. As I said, I *never* saw a simulator with this option.

There are two concepts, and some confusion. one concept is "the current
is limited", and the other is "you should not operate beyond that
limit". I don't know which one is appopriate for NPN models, but from
the OP, my impression was that the model should have or could have
limited the current.

> It would be awkward anyway; the 100 mA maximum is not a device hard limit

Thanks for the clarification, so we probably are in the second case.
This is standardised in Verilog-A (through $warning), and I bet somebody
has used it. (gnucap-adms supports it).

We still need a netlist (ideally minimal) to understand the original
issue. I remember there was a switch involved...

Rmano, 2020-10-15:
> There are two concepts, and some confusion. one concept is "the current is limited", and the other is "you should not operate beyond that limit".

Ah yes! Now I understand, no, for the BC547 is clearly the second one. The only limitation in the device is the magic smoke.