Qucs/qucs #1017: Component types (sources, probes, ...) missing from drop-down

xy1xy, 2020-10-22:
I have installed qucs (v0.0.20) via the AUR package qucs on Manjaro. And I've also installed the git version which has the same issue described below.

The only component types available is "lumped components"
[image]

When I install the qucs-spice (v0.0.19) package from SNAP I get a lot more options:
[image]

I really need access to the sources and probe components - which are not available.

NOTE: sorry about the quality of the screenshots, but when the drop-down component list is expanded, it seems to capture all keystrokes, so my screenshot tool (FlameShot) doesn't work.

Rmano, 2020-10-22:
Which version are you using (compiled, binary,...), and on which OS? My 0.0.20-rc2 (the [AppImage version I built](https://rmano.github.io/qucsAppImagesBuild/), which is the one that currently runs on my Ubuntu 20.04 ) works ok:

[image]

felix-salfelder, 2020-10-22:
On Wed, Oct 21, 2020 at 05:34:13PM -0700, xy1xy wrote:
> I have installed qucs (v0.0.20) via the AUR package qucs on Manjaro.
> And I've also installed the git version which has the same issue
> described below.

Recently, somebody reported that the AUR package (possibly
0.0.20_rc2-2?) should work [1], but no details. You could try to give
more details about your own installation, and especially about your Qt.

> NOTE: sorry about the quality of the screenshots [..]

have you tried scrot? note that screenshots are not required, as your
description is perfectly clear.

[1] https://aur.archlinux.org/packages/qucs/#comment-770535

Rmano, 2020-10-22:
> have you tried scrot? note that screenshots are not required, as your description is perfectly clear.

In defense of the OP, is it true that the drop-downs (but I think it's not qucs, it's Qt doing that) are very aggressive in grabbing keys --- not even the Expose key works when displayed. I personally use[ `ksnip` ](https://github.com/ksnip/ksnip)with a delay...

xy1xy, 2020-10-22:
Thanks for your replies, I had forgotten about applying a delay for the screenshot in this instance. Worked with Flameshot:

[image]

I've tried all 3 Arch linux AUR packages, all with the same result (only lumped components available)
https://aur.archlinux.org/packages/?O=0&K=qucs

All 3 packages have a dependency on qt4 (4.8.7)

And the source is built using these sources:

- **qucs:** http://downloads.sourceforge.net/project/qucs/qucs/0.0.20/qucs-0.0.20-rc2.tar.gz
- **qucs-git:** git+https://github.com/qucs/qucs
- **qucs-s:** https://github.com/ra3xdh/qucs_s/releases/download/0.0.22/qucs-s-0.0.22.tar.gz

rfbug, 2021-03-21:
Same here... compiling from the current develop branch yields only 'lumped components' in the 'Components' tab

@Rmano I suspect something must have broken after the time of build of your AppImage, since I also have a succesful '0.20'-build (fd5a8e3f)...

felix-salfelder, 2024-07-06:
It looks okay after the Qt5 move (see develop branch). Can't tell what the source of the problem is. Please reopen if this is still relevant.

closed 2024-07-06 by felix-salfelder