Qucs/qucs #1019: Problems building refactor+gt5-19 branch

wkazubski, 2020-11-28:
I am trying to build latest qt5 branch of QUCS.This is due to Qt4 being phased out from openSUSE Tumbleweed.
So far I noticed two problems, the first one is caused by bootstrap file listing qucs-core as one of modules to build (line 9) but corresponding subdirectory was removed some time ago.
The second problem is caused by build failng to find the libqucsator library. The source code for this library is not included, so I build it (as distribution package) from qucsator-0.0.20 tarball. After that the build enters empty qucsator subdirectory and fails on missing Makefile (directory is empty).

felix-salfelder, 2020-11-28:
thanks for the useful feedback.

On Sat, Nov 28, 2020 at 02:20:07PM -0800, wkazubski wrote:
> So far I noticed two problems, the first one is caused by bootstrap
> file listing qucs-core as one of modules to build (line 9) but
> corresponding subdirectory was removed some time ago.

right, this bug was also present in develop. will be part of
refactor+gt5-20 (i.e. after the next rebase).

> The second problem is caused by build failng to find the libqucsator
> library. The source code for this library is not included, so I build
> it (as distribution package) from qucsator-0.0.20 tarball. [...]

Not exactly sure where this happens, as I could not find a reference to
libqucsator. Using qucsator from the release tarball is the correct way
forward.

The files outside of $git_root/qucs/ need work. I need to find the right
moment to split it up. Maybe just before the refactored code becomes
usable, or after a 0.0.20 release. For now, please ignore it, i.e. cd
$git_root/qucs; ./bootstrap etc.

wkazubski, 2020-11-29:
This is  the output of libtool, the last part of the build log:
...
libtool: link: g++  -fPIC -DPIC -shared -nostdlib /usr/lib64/gcc/x86_64-suse-linux/10/../../../../lib64/crti.o /usr/lib64/gcc/x86_64-suse-linux/10/crtbeginS.o  .libs/legacylib_la-libfiles.o   -ldl -L/usr/lib64/gcc/x86_64-suse-linux/10 -L/usr/lib64/gcc/x86_64-suse-linux/10/../../../../lib64 -L/lib/../lib64 -L/usr/lib/../lib64 -L/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/lib -L/usr/lib64/gcc/x86_64-suse-linux/10/../../.. -lstdc++ -lm -lc -lgcc_s /usr/lib64/gcc/x86_64-suse-linux/10/crtendS.o /usr/lib64/gcc/x86_64-suse-linux/10/../../../../lib64/crtn.o  -O2 -g -m64 -fstack-protector -O2 -g -m64 -fstack-protector   -Wl,-soname -Wl,legacylib.so -o .libs/legacylib.so
libtool: link: ar cru .libs/legacylib.a  legacylib_la-libfiles.o
libtool: link: ranlib .libs/legacylib.a
libtool: link: ( cd ".libs" && rm -f "legacylib.la" && ln -s "../legacylib.la" "legacylib.la" )
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I.. -I/usr/include/qt5/QtCore -I/usr/include/qt5 -I/usr/include/qt5/QtGui -DQT_SCRIPT_LIB -I/usr/include/qt5/QtScript -DQT_SVG_LIB -I/usr/include/qt5/QtSvg -DQT_WIDGETS_LIB -I/usr/include/qt5/QtWidgets -DQT_GUI_LIB -DQT_XML_LIB -I/usr/include/qt5/QtXml -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_DEBUG_OUTPUT -DQT_DEPRECATED_WARNINGS -DQT_THREAD_SUPPORT -D_REENTRANT -fPIC -DQT_PRINTSUPPORT_LIB -I/usr/include/qt5/QtPrintSupport -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I/usr/include/qt5/QtGui -DQT_WIDGETS_LIB -I/usr/include/qt5/QtWidgets -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SCRIPT_LIB -I/usr/include/qt5/QtScript -I/usr/include/qt5 -DQT_CORE_LIB -I/usr/include/qt5/QtCore -I../include -I../qucs -O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -pipe -fno-check-new -std=c++14 -fPIC -O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -pipe -fno-check-new -std=c++14 -c qucsator.cpp -o qucsator_la-qucsator.o >/dev/null 2>&1
/bin/sh ../libtool  --tag=CXX   --mode=link g++ -O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -pipe -fno-check-new -std=c++14 -fPIC -O2 -g -m64 -fmessage-length=0 -D_FORTIFY_SOURCE=2 -fstack-protector -funwind-tables -fasynchronous-unwind-tables -pipe -fno-check-new -std=c++14 -module -avoid-version  -o qucsator.la -rpath /usr/lib64/qucs qucsator_la-qucsator.lo qucsator_la-dat.lo qucsator_la-qucsator.moc.lo  -ldl 
libtool: link: g++  -fPIC -DPIC -shared -nostdlib /usr/lib64/gcc/x86_64-suse-linux/10/../../../../lib64/crti.o /usr/lib64/gcc/x86_64-suse-linux/10/crtbeginS.o  .libs/qucsator_la-qucsator.o .libs/qucsator_la-dat.o .libs/qucsator_la-qucsator.moc.o   -ldl -L/usr/lib64/gcc/x86_64-suse-linux/10 -L/usr/lib64/gcc/x86_64-suse-linux/10/../../../../lib64 -L/lib/../lib64 -L/usr/lib/../lib64 -L/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/lib -L/usr/lib64/gcc/x86_64-suse-linux/10/../../.. -lstdc++ -lm -lc -lgcc_s /usr/lib64/gcc/x86_64-suse-linux/10/crtendS.o /usr/lib64/gcc/x86_64-suse-linux/10/../../../../lib64/crtn.o  -O2 -g -m64 -fstack-protector -O2 -g -m64 -fstack-protector   -Wl,-soname -Wl,qucsator.so -o .libs/qucsator.so
libtool: link: ar cru .libs/qucsator.a  qucsator_la-qucsator.o qucsator_la-dat.o qucsator_la-qucsator.moc.o
libtool: link: ranlib .libs/qucsator.a
libtool: link: ( cd ".libs" && rm -f "qucsator.la" && ln -s "../qucsator.la" "qucsator.la" )
make[3]: Leaving directory '/home/wk/rpmbuild/BUILD/qucs/qucs/legacy'
make[3]: Entering directory '/home/wk/rpmbuild/BUILD/qucs/qucs'
make[3]: Leaving directory '/home/wk/rpmbuild/BUILD/qucs/qucs'
make[2]: Leaving directory '/home/wk/rpmbuild/BUILD/qucs/qucs'
make[1]: Leaving directory '/home/wk/rpmbuild/BUILD/qucs/qucs'
Making all in qucsator
make[1]: Entering directory '/home/wk/rpmbuild/BUILD/qucs/qucsator'
make[1]: *** No rule to make target 'all'.  Stop.
make[1]: Leaving directory '/home/wk/rpmbuild/BUILD/qucs/qucsator'
make: *** [Makefile:381: all-recursive] Error 1

maybe this will hep.

wkazubski, 2020-11-29:
I removed qucsator from AC_CONFIG_SUBDIRS in lines 48/49 of configure.ac and from SUBDIRS in line 20 of Makefile.am.
Now build goes trough.

I suspect that cucsator foler in your bul tree may contain some old files that were excluded from current git.
Wojciech

felix-salfelder, 2020-11-29:
On Sun, Nov 29, 2020 at 06:51:35AM -0800, wkazubski wrote:
> I suspect that cucsator foler in your bul tree may contain some old files that were excluded from current git.

The qucsator folder is a git submodule, and correctly appears as an
empty directory in a fresh clone.  c.f. README.md for instructions and
explanation of how this is meant to work. if you don't need qusator from
here, do "cd qucs" before you build. see my previous reply.

closed 2020-12-22 by felix-salfelder