Qucs/qucs #1020: verilog file bloc doesn't handle parameter correctly

Patarimi, 2020-11-30:
I write the code below in verilog:
[spi.txt](https://github.com/Qucs/qucs/files/5615133/spi.txt)
Then I imported it in qucs using the verilog file bloc, I was expecting the left bloc and I get the bloc on the right, is it the expected behaviour ?
[image]

PS: I use qucs 0.0.19 on windows 10.

felix-salfelder, 2020-11-30:
On Mon, Nov 30, 2020 at 02:00:41AM -0800, Patarimi wrote:
> I write the code below in verilog:
> module spi_slave
>     #(
>     parameter WORD_WIDTH = 8
>     ) (
>     //spi interface
>     input mosi,
>     input sck,
>     input ssn,

the syntax looks odd to me. if it is verilog, then maybe it is an unsupported
version?  (how do parameters look like in files that work?)

Patarimi, 2020-12-01:
I use this [tutorial](https://www.chipverify.com/verilog/verilog-parameters). It is probably something added with [verilog 2001](https://www.chipverify.com/images/verilog/verilog_keywords.png).
> how do parameters look like in files that work?

Do you mean, how I get the left symbol in the picture ? I simply remove the parameter section (the parameter WORD_WIDTH is undefined).

felix-salfelder, 2020-12-01:
On Mon, Nov 30, 2020 at 11:13:54PM -0800, Patarimi wrote:
> Do you mean, how I get the left symbol in the picture ? I simply
> remove the parameter section (the parameter WORD_WIDTH is undefined).

no. I mean, have you seen a parameter working at all, e.g. with a
different syntax, in an unrelated module, in a different file?

Patarimi, 2020-12-02:
No, but it change to blue in the text editor. I though it was handled.

felix-salfelder, 2020-12-02:
On Wed, Dec 02, 2020 at 12:34:11AM -0800, Patarimi wrote:
> No, but it change to blue in the text editor. I though it was handled.

which other tool supports this syntax (in any)?

Patarimi, 2020-12-02:
> which other tool supports this syntax (in any)?

[Vivado 2020](https://www.xilinx.com/products/design-tools/vivado.html) (from Xilinx) supports it.

felix-salfelder, 2020-12-02:
On Wed, Dec 02, 2020 at 03:00:49AM -0800, Patarimi wrote:
> [product] supports it.

I am surprised. Does it synthesise such text, or does it just accept it?
Unlike Qucs, commercial tools sometimes try to read your mind..

The verilog-ams lrm [1] suggests that module parameters are declared in
the module body. This is my interpretation, and it may be wrong. Note
that I did not implement the verilog parser in Qucs, nor have I read it.

hth

[1] https://www.accellera.org/images/downloads/standards/v-ams/VAMS-LRM-2-4.pdf

Patarimi, 2020-12-02:
Yes, the synthesis and implementation is successful using Vivado. I use this syntax to be able to parametrize the buses width.
In the document you linked, I think the syntax can be seen in the part A.1.3.

I am new to QUCS and I may have also miss something.
Thanks for your time.

felix-salfelder, 2020-12-02:
On Wed, Dec 02, 2020 at 05:17:56AM -0800, Patarimi wrote:
> In the document you linked, I think the syntax can be seen in the part A.1.3.

Okay, I can see it now. there is a module_parameter_port_list, also
mentioned in 6.2.

It seems, I was looking at 2.8 instead. block_item_declaration also
matches a parameter_declaration, and this is what I had in mind.

Curiously, all examples in the lrm use the latter way of declaring
parameters, and there is no motivation or rationale given for the
module_parameter_port_list.

> I am new to QUCS and I may have also miss something.

Clearly, I have missed something. Maybe you could try the other way?
I'd be happy to learn if it does (not) do what you need. Or what the
difference is, to begin with.

thanks!

PS: Theres refactoring work to do in the verilog area, maybe this would
not be hard to add and should be considered eventually. looking for
volunteers...

Patarimi, 2020-12-02:
> I'd be happy to learn if it does (not) do what you need. Or what the
difference is, to begin with.

The way I use it is to parametrize the port `output reg [WORD_WIDTH-1:0] rword,`. If you declare the parameter after the port part, the parameter is not yet declared (I think this is expected).

> PS: Theres refactoring work to do in the verilog area, maybe this would
not be hard to add and should be considered eventually. looking for
volunteers...

I would help with great pleasure. I never develop something like that before, is there a guide on how to get started ?

felix-salfelder, 2020-12-02:
On Wed, Dec 02, 2020 at 06:20:25AM -0800, Patarimi wrote:
> The way I use it is to parametrize the port `output reg
> [WORD_WIDTH-1:0] rword,`. If you declare the parameter after the port
> part, the parameter is not yet declared (I think this is expected).

Also, you just pointed me to the solution of a problem I had yesterday.
Well done.

> I would help with great pleasure. I never develop something like that
> before, is there a guide on how to get started ?

Not a guide really, but guidance. Please email, either to me or to
qucs-devel.

Patarimi, 2020-12-11:
Glad to have help, even if it was by luck :-)
I have started to look into this. Should this issue be transferred to qucs/ADMS ?

felix-salfelder, 2020-12-11:
On Fri, Dec 11, 2020 at 01:00:27AM -0800, Patarimi wrote:
> Glad to have help, even if it was by luck :-)
> I have started to look into this. Should this issue be transferred to qucs/ADMS ?

We need a component that takes the name of a verilog file and a module
name and expands to a symbol with the ports and parameters as in that
module. The drawing should be the box we see in your screenshots.

Next, we need a variation/extension of this. It also takes a reference
to a different drawing, so it does not always look like a box. Such a
thing could be conflated in a .lib file, but other ways are feasible.

While the purpose of plugins is not to interfere with your own ideas of
doing things, I strongly advise not to use ADMS here. There are better
ways, and there are much better ways...

NB: This is about the "refactored" work-in-progress. I don't know what
is achieveable in the current stable/develop branches, nobody is
maintaining that.