Qucs/qucs #1030: ASCO with Qucs 0.0.19

GMellar, 2020-12-30:
I tried the optimization for microstrip filters but the ASCO (0.4.10) output is not read at all due to the wrong filename. Qucs sets the output file as parameter but that isn't read after generation because ASCO uses the hostname as default output log file. 

I looked into the source of Qucs and the output_log filename is always asco_out.log. A simple hack for me to output the log to a fixed filename worked for me. I think qucs is a great tool. In this case I would recommend to either include a modified ASCO version inside the source tree or parse the hostname based file which is the default output of ASCO. 

It is also quite inconvenient to start the optimization, wait and don't see the optimization result in a last simulation step. What I need to do is start the optimization, go into the optimization dialog, copy the variables, paste them into the sheet and disable the optimization. After another run I see the result. I think it could be done better by parsing the output log (which already happens) and then replace the variables inside the sheet and run another simulation without the optimization. This would be much more user friendly.

felix-salfelder, 2020-12-30:
On Wed, Dec 30, 2020 at 03:07:11AM -0800, Geoffrey Mellar wrote:
> ... include a modified ASCO version inside the
> source tree or parse the hostname based file which is the default
> output of ASCO. 

Please see the qucs-pkg repo. There is an ASCO patch available in the
contrib section. I could not find any install script or documentation
though, and I did not try to find an upstream. Looks like it needs work.

> It is also quite inconvenient to start the optimization, wait and
> don't see the optimization result in a last simulation step. What I
> need [..]

In future Qucs (past 0.0.20) such things will be plugins. Somebody will
have to to port or implement an ASCO (or similar) driver to run these
optimisations.

sv1, 2021-05-08:
I am using qucs 0.0.20 and have applied the asco patch found on contrib section but I still have problem:

Starting new simulation on Σάβ 08. Μαϊ 2021 at 11:26:08:015

creating netlist... done.
Starting /opt/linux/ASCO-0.4.10/asco

ASCO-0.4.10 - Copyright (c) 1999-2016 Joao Ramos
This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU General Public License for more details.

INFO:  Qucs initialization on 'apollo'
INFO:  Initialization has finished without errors on 'apollo'
INFO:  Current directory on 'apollo': /home/akefalas/.qucs

                         PRESS CTRL-C TO ABORT

INFO:  Starting global optimizer on 'apollo'...
       ...using local-global optimizer for: Parameter 1 Parameter 2 Parameter 3
auxfunc_measurefromlis.c - ProcessOutputFile -- Cannot open output file: apollo.dat

Errors occurred during simulation on Σάβ 08. Μαϊ 2021 at 11:26:08:084
Aborted.