Qucs/qucs #1037: Qucs-S

cederom, 2021-02-09:
Just found amazing Qucs-S [1] that adds NGSPICE to Qucs.. looks like more and more folks are waiting for QT5/QT6 update :-)

[1] https://github.com/ra3xdh/qucs_s

felix-salfelder, 2021-02-09:
On Tue, Feb 09, 2021 at 06:06:33AM -0800, Tomasz CEDRO wrote:
> Just found amazing Qucs-S [1] that adds NGSPICE to Qucs.. looks like
> more and more folks are waiting for QT5/QT6 update :-)

Hi Tomasz.

We essentially need a netlister to run NGSPICE (or any spice, for that
matter). There is no need to wait for the QT5/QT6 update.

Let me know if you'd like to help (email).

cederom, 2021-02-09:
Thanks @felix-salfelder for quick response! How about integrating QUCS-S into KiCAD? That would be the biggest benefit for everyone..? :-)

felix-salfelder, 2021-02-09:
On Tue, Feb 09, 2021 at 06:42:14AM -0800, Tomasz CEDRO wrote:
> Thanks @felix-salfelder for quick response! How about integrating
> QUCS-S into KiCAD? That would be the biggest benefit for everyone..?
> :-)

I am not going to stop you.

Intrgrating Qucs-S into Qucs is already in progress.

(For conversation, please use the mailing list, not the issue tracker).

cederom, 2021-02-09:
Great news :-) This is my first contact with these project. Thank you, you are doing amazing job! :-)

closed 2021-02-09 by cederom