Qucs/qucs #1042: simulation crushes

Abbode, 2021-04-21:
hello everyone , I made a simple circuit as to try qucs , I used what I learnt in the docs , yet no simulation would run .
this is the netlist : 

R:R1 Vout _net0 R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
R:R2 gnd Vout R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
Vdc:V1 _net0 gnd U="1 V"
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"

this is the log file : 

[log.txt](https://github.com/Qucs/qucs/files/6352949/log.txt)

I'm using QUCS 0.0.19 on fedora workstation 33 with GNOME UI .

thanks are in advance..

felix-salfelder, 2021-04-21:
On Wed, Apr 21, 2021 at 10:52:34AM -0700, Abd AlHaleem Bakkor Mustafa wrote:
> this is the netlist :
> R:R1 Vout _net0 R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
> R:R2 gnd Vout R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
> Vdc:V1 _net0 gnd U="1 V" [...]

> I'm using QUCS 0.0.19 on fedora workstation 33 with GNOME UI .

Your netlist appears to work for me with qucsator 0.0.20, in

$ qucsator -i netlist_file

There is some indication that you are using qucsator 0.0.19, and a newer
one may be worth a try.

Abbode, 2021-04-21:
> On Wed, Apr 21, 2021 at 10:52:34AM -0700, Abd AlHaleem Bakkor Mustafa wrote: this is the netlist : R:R1 Vout _net0 R="50 Ohm" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85" R:R2 gnd Vout R="50 Ohm" [...]

this is the latest version in fedora's repositories .... what can I do to update ?

felix-salfelder, 2021-04-22:
On Wed, Apr 21, 2021 at 02:04:33PM -0700, Abd AlHaleem Bakkor Mustafa wrote:
> this is the latest version in fedora's repositories .... what can I do to update ?

I have no control over fedora. Such things usually take quite a while.
But it should be straightforward to build Qucsator 0.0.20.

Abbode, 2021-04-22:
> On Wed, Apr 21, 2021 at 02:04:33PM -0700, Abd AlHaleem Bakkor Mustafa wrote: this is the latest version in fedora's repositories .... what can I do to update ?
> I have no control over fedora. Such things usually take quite a while. But it should be straightforward to build Qucsator 0.0.20.

sure you don't , but developers do have control over distributing their applications and they should keep pushing updates for them  .
either way I think I'll try a different packaging system before trying building it myself  , because keep dealing with builds to keep it up-to-date is the last thing I want to do .

felix-salfelder, 2021-04-22:
On Thu, Apr 22, 2021 at 01:56:35AM -0700, Abd AlHaleem Bakkor Mustafa wrote:
> sure you don't,

I don't know what you mean by that.

> but developers do have control over distributing
> their applications and they should keep pushing updates for them  .

Qucsator is not currently maintained by anybody. Nor developed. But
there is a release tarball. I'm almost certain it will work on fedora.

> either way I think I'll try a different packaging system before trying
> building it myself

There is a Qucsator package in the Arch User Repository, some kind of
inofficial collection for Arch Linux. I don't know if installing that
package qualifies as "building it [your]self", but it may be worth a
try.