Qucs/qucs #1044: v0.0.20 ADMS compile difficulties

ma-laforge, 2021-06-09:
I created a simple VerilogA `V(out, xGND) <+ abs(V(in, xGND))` test block in 2015 that worked magnificently in March 2015 (whatever version of Qucs was available then).

I wanted to experiment further with Qucs' VerilogA capabilities - so I installed v0.0.20 using `apt`, as outlined in
-->http://qucs.sourceforge.net/install.html.

...but I hit a few walls:
 1. Opened my `V_absolute.va` verilog-A model
 1. Clicked `Project|Build Verilog-A module...`
 1. Noticed the "make" statement in admsXml Dock:
```
make -f /usr/include/qucs-core/va2cpp.makefile ADMSXML=/usr/bin/admsXml PREFIX=/usr MODEL=V_absolute
```
 4. But wait! Qucs did not install `/usr/include/qucs-core`!
 1. Ok, after reading a bit, I found that `qucs-core` was renamed --> `qucsator`.
 1. Ok!? so I run `sudo ln -s qucsator qucs-core`
 1. Awesome! simulation starts... but now I get a message that `disciplines.vams` cannot be found.
 1. ...Neither can I!?! Plus my version of admsXml is old (circa 2015!?)
 1. So I download the latest release of ADMS (v2.3.7).
 1. And I install from tarball as explained in the `README`:
```
tar xvfz adms-x.x.x.tar.gz
cd adms-x.x.x
./configure --prefix=/usr
make install
```
 11. Awesome! I now have a `disciplines.vams` & `constants.vams` in `/usr/include/adms`
 1. I also have a recent version of `admsXms` now!

### I think there are issues
with the ADMS distribution bundled with Qucs. It was sort of tough to get up and running.

felix-salfelder, 2021-06-09:
On Wed, Jun 09, 2021 at 09:01:24AM -0700, ma-laforge wrote:
> I wanted to experiment further with Qucs' VerilogA capabilities - so I installed v0.0.20 using `apt`, as outlined in
> -->http://qucs.sourceforge.net/install.html.

Thanks for your feedback.

There is no "v0.0.20" (yet), and you might have installed an unofficial
package. I can see references to ppa:qucs/qucs in install.html, did you
use this (and what is it)?

> ### I think there are issues
> with the ADMS distribution bundled with Qucs. It was sort of tough to get up and running.

The develop branch does no longer bundle anything. There is a qucs-pkg
repo for the purpose of bundling. It is neither complete, nor has it
ever been released. There is some work to do... I believe you found an
inconsistency that could be fixed (some broken reference to qucsator
files).

You could also try the 0.0.20-rc2 snapshot.

ma-laforge, 2021-06-10:
Apologies if I don't answer all your questions correctly. I am not 100% certain what you are asking me to do.

### My installation
> I can see references to ppa:qucs/qucs in install.html, did you use this (and what is it)?

Yes. I applied the procedure that appeared to be the most common with Ubuntu installations I have done in the past:
```
sudo apt-add-repository ppa:qucs/qucs
sudo apt-get update
sudo apt-get install qucs
```

I am using Ubuntu 18.04 (64-bit), so it installed the following version (screen grab from Synaptic package manager):
[image]

### Better installation method
> You could also try the 0.0.20-rc2 snapshot.

I got something to work at the moment, so I'll stick with it for now. But out of curiosity, if I ***were*** to install it, would I apply the following procedure?:
```
tar xvfz qucs[version].tar.gz
cd qucs[version]
./configure
make install
```

(using the tarball in the [releases](https://github.com/Qucs/qucs/releases) section)?

felix-salfelder, 2021-06-10:
On Wed, Jun 09, 2021 at 07:01:35PM -0700, ma-laforge wrote:
> I got something to work at the moment, so I'll stick with it for now.
> But out of curiosity, if I ***were*** to install it, would I apply the
> following procedure?:
> ```
> tar xvfz qucs[version].tar.gz
> cd qucs[version]
> ./configure
> make install
> ```

Yes. Tags in the git repo should correspond to the tarball version, so
you could obtain the code from git as well. But bootstrapping is
required, which has more dependencies. Results may vary.

You have noticed that Qucs is not part of Debian/Ubuntu, or any other
major distro. The reason is that Qt<=4 has been phased out. Packages
from third parties such as ppa do exist and involve workarounds, but may
be less reliable.

felix-salfelder, 2021-07-15:
Please reopen if you have further questions or refer to the other tickets about 0.0.20.

closed 2021-07-15 by felix-salfelder