Qucs/qucs #1045: Have you considered replacing Octave with Julia?

ma-laforge, 2021-06-10:
I like what the team has done with this project - and I definitely like the idea of using a mathematical language like Octave.

But my experience is that Julia is more stable/predictable/reliable/etc than Octave.  Not to mention that it is natively fast once things are compiled.

I've noticed that a few designers are at least aware of the language:
 - https://github.com/Qucs/qucs/issues/563#issuecomment-261513323
 - https://github.com/Qucs/qucs/issues/166#issuecomment-69332246

Anyways: just putting it out there.

felix-salfelder, 2021-06-10:
On Wed, Jun 09, 2021 at 07:08:04PM -0700, ma-laforge wrote:
> But my experience is that Julia is more stable/predictable/reliable/etc than Octave.

You seem to imply that Octave should no longer be used because now there
is Julia. There are many other scripting and programming languages.

I do not believe "replace" is the correct approach. Maybe you could run
julia instead of octave if your analysis happens to be implemented in
julia.

Please use the mailing list for this kind of discussion.

ma-laforge, 2021-06-10:
Apologies. I have never used mailing lists before. I was unaware you preferred other more modes of communications for these types of discussions.

> I do not believe "replace" is the correct approach.
That's fair. I did not mean to sidetrack your development.

closed 2021-06-10 by ma-laforge