Qucs/qucs #1047: Got trouble in ./bootstrap

abdulbadii, 2021-07-13:
Got interrupted when follow build steps:
```
$ ./bootstrap
Running aclocal...
Running automake...
Running autoconf...
bootstrapping /home/budi/Downloads/qucs/qucsator
./bootstrap: 28: ./bootstrap: ./bootstrap: not found
trouble in qucsator/bootstrap

```

How to correctly solve it?

felix-salfelder, 2021-07-13:
On Mon, Jul 12, 2021 at 05:32:13PM -0700, abdulbadii wrote:
> ```
> [..]
> trouble in qucsator/bootstrap
> ```
> 
> How to correctly solve it?

Thanks for raising this. I assume that you have checked out the develop
branch in the "qucs" repo (29432325df).

A possible way forward would be to drop all but qucs/ subdirectory
contents, and move the files to top. (NB: Doing so will disrupt the
0.0.20 release). After this step, the "qucs-pkg" repo can be set up to
provide what should have gone into the 0.0.20 release.

In the meantime... README.md could be more explicit about cloning the
submodules. Ideally, bootstrap would produde a clearer error message
and/or point to instructions.

felix-salfelder, 2021-07-15:
> In the meantime [..]

done, in develop. Please reopen if it does not solve the issue.

closed 2021-07-15 by felix-salfelder