Qucs/qucs #1048: Qucs with Subcircuits and ASCO

cweickhmann, 2021-07-29:
With the
* Windows version of Qucs 0.0.19 (win32, mingw482) and ASCO 0.4.10, as well as with
* Linux version of Qucs 0.0.20 () I ran into a rather strange problem.

Premise:
* A Project
* One circuit file with a subcircuit to optimize
* The subcircuit file

The subcircuit, say, has an element "L1" with the value given as "ind1". An equation is given in the subcircuit, where
"ind1" is assigned the parameter "Lcorner".
The circuit file, an equation is given with a parameter "Lc" which is the variable to optimise.  I assign "Lc" to "Lcorner" in the subcircuit block.
This runs fine without the optimiser block (e.g. in an S-parameter simulation).

Once you activate the optimiser block (with that variable and goals defined), it fails with a syntax error like "line 4: syntax error, unexpected Identifier".
When you open the `netlist.txt` file, you won't find a syntax error in line 4. Everything is fine there.

However, if you open `asco_netlist.txt`, you'll find something strange happening to the Eqn:Eqn1 line in the subcircuit block.
I found two variations of this. Either

```
.Def:subcircuit _net0 _net1 Lcorner="1n"
L:L1 _net0 _net1 L="ind1" I=""
Eqn:Eqn1 ind1="#Lc#orner" Export="yes"
.Def:End
```
(note the `#Lc#` replacement)

OR
```
.Def:subcircuit _net0 _net1 Lcorner="1n"
L:L1 _net0 _net1 L="ind1" I=""
Eqn:Eqn1 ind1="1.141129E-9orner" Export="yes"
.Def:End
```
(note the literal replacement of "Lc" with its numerical value.

I am not sure this is a bug in Qucs or ASCO, but it is certainly irritating. Especially, since it seems to be caused simply by using an equation in the subcircuit.

You can work around this, if you don't allow variables to start with the same characters. I.e. in this case, the problem is solved by calling `Lc` `L_c` instead.
[test_mwe_prj.zip](https://github.com/Qucs/qucs/files/6899483/test_mwe_prj.zip)

ckoegler, 2021-08-03:
Hi,
I can reproduce it with qucs 0.0.20-rc2 and ASCO-0.4.10.
I only get the version with
      ind1="#Lc#orner"
How did you provoke 
      ind1="1.141129E-9orner"
?
It's most probably caused by qucs. If I reneame the asco executable, qucs complains that it cannot find asco, but still generates the fawlty netlist.

cweickhmann, 2021-08-05:
Hi! Thanks for picking this up!

I tried to reproduce the float replacement, but did not succeed yet. I hope I can tell you soon.
It happened in an optimisation with s-parameter sim and an touchstone block, if that's of any help.

ckoegler, 2021-08-05:
Don't worry, its not so important. The float replacement is done probably by asco, following a fawlty replacement from qucs.

note:
The asco documentation says about the spice input netlist:

> All values that are to be replace by the optimizer are enclosed in number sign # #.
> This does not necessarily imply a variable to optimize. It can also be a fixed number
> whose value is set for flexibility in the configuration file. This is a simple method
> to optimize a write protected SPICE input netlist circuit with different operating
> conditions.