Qucs/qucs #1050: Windows MSYS2 build - qt4 removed?

ddrake, 2021-09-06:
Hi, I was following the instructions in the wiki: https://github.com/Qucs/qucs/wiki/Build-Windows using MSYS2 on a Windows 10 machine.  When this command failed: `pacman --noconfirm -S mingw-w64-x86_64-qt4`, I looked at the .appveyor.yml file and saw the links and instructions for the download and offline installation work-around.  But `wget -nc http://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-qt4-4.8.7-4-any.pkg.tar.xz` failed so I checked the directory http://repo.msys2.org/mingw/x86_64/ and no qt4 package seems to exist there anymore.  Has anyone been able to find an alternative repo?

neilharvey94044, 2022-01-23:
I'm running into problems with the Windows 10 build as well.  For the specific problem you identify I tried using qt5 and that got me through the step.
I have lots of other remaining issues with the build, however.  These instructions may be outdated.

felix-salfelder, 2024-07-02:
Situation has changed with qt5 in the develop branch.

closed 2024-07-02 by felix-salfelder