Qucs/qucs #1054: QUCS nomenclature for "Relays" in the ".sch" file is "Relais" - Typo?

Circa-Tree, 2021-09-17:
### "Relays" in the schematic diagram are called "Relais" in the ".sch" file - Typo?

Relays in the schematic diagram are called "Relais" in the .sch file, which is not consistent naming, and is probably a typo.
Not helpful, when working at the sch file level (using text editor).

Relais in French can mean hotel, or staging post, but when using the english version of the program, it is expectable that the english word (in the .sch file) is used to represent the english component name.

Circa-Tree, 2021-10-22:
Looked for "modular QUCS" and only found [QUC-S](https://github.com/Qucs/qucs-s) which seems to a module for use in "modular QUCS".

Is there a link for "modular QUCS" ?
If it is possible to install modular QUCS along side original QUCS, without causing problems then it can be conveniently trialled.

Not seen a reference to localization of QUCS, but there are users from a diversity of languages, who probably represent a small proportion of potential users in each of those languages.

Alias component names will be great (as might other parameters) as the analysis engine can be used for many different 'systems' (other than electronic, especially if it is possible to easily switch between alias modes/languages/'field of study'.

felix-salfelder, 2025-01-18:
Legacy device names must stay as they are for compatibility. Nothing we can do here.

closed 2025-01-18 by felix-salfelder