Qucs/qucs #1061: Compile linking dropped libraries statically.

emerajid, 2021-11-10:
Hi guys.
You know that with qt4 drop on many distributions it is keen pain in known place to get qucs working and I don't see anything as good as it in open source world. Moreover, qucs seems to be the only option that satisfies my needs, so I badly need it. And it is quite stupid that I have to use qucs under wine.
So my question is:
Is there anyone who could compile qucs for linux linking at least dropped libraries statically? I don't care about size, I just want qucs to run natively. Or it is not done so easily?
I ask because I couldn't neither compile qucs on debian 11, nor get already compiled binary to work well (it seems that simulation engine crashes for no reason, though in debian 11.0 it worked; I think most recent point release broke the stuff). Appimage does not work for me for some reason. Snap and flatpak are out of question, because I'd rather use wine.
Also, I think it must be easier than porting everything to qt5, so many people, who got used to qucs, would appreciate if you did so.

felix-salfelder, 2021-11-10:
> [..] (it seems that simulation engine crashes for no reason, though in
> debian 11.0 it worked; I think most recent point release broke the
> stuff)

One crash is fixed with commit e69c127be6c84 earlier this year. There
hasnt been a release since. Try Qucsator from the develop branch.

If there are other crashes, please report them independently, and in
more detail.

emerajid, 2021-11-10:
> [..] (it seems that simulation engine crashes for no reason, though in debian 11.0 it worked; I think most recent point release broke the stuff)
> One crash is fixed with commit e69c127be6c84 earlier [...]

There is much trouble for me about compiling Qucsator. For some reason compiler throws error at certain point. There is much trouble also about compiling a GUI (or getting a precompiled one). I though that it would be much better if developers themselves produced binary with everything bundled in and people could benefit from it.
Of course I could spend time figuring why it won't compile, but not that I have much time, that's why I'm asking people, who devote their time to Qucs.

felix-salfelder, 2021-11-10:
On Wed, Nov 10, 2021 at 08:57:53AM -0800, emerajid wrote:
> There is much trouble for me about compiling Qucsator. For some reason
> compiler throws error at certain point.

Please share more details about what the trouble is exactly. This is
essential to anybody who might be willing to fix it. Please open a new
thread about the specific issue.

> would be much better if developers themselves produced binary with
> everything bundled in and people could benefit from it.

This is true. Yet, there is no developer.

If you would like to share Qucs/Qucsator binaries, you are not alone.
I have seen a related post here [1] just hours ago.

> Of course I could spend time figuring why it won't compile, but not
> that I have much time, that's why I'm asking people, who devote their
> time to Qucs.

It would indeed be great if these people existed. If you need help
figuring out stuff, feel free to ask.

[1] https://alioth-lists.debian.net/pipermail/pkg-electronics-devel/2021-November/008248.html

emerajid, 2021-11-13:
> Yet, there is no developer.
> if these people existed.
Do you mean that things are really bad and there are very few (or none at all) people who spend their time developing qucs? If so, this is really sad and I wish I could help you
> If you would like to share Qucs/Qucsator binaries, you are not alone.
but I am not likely to learn programming in the nearest years... Especially such hard and complex languages as C++.

felix-salfelder, 2021-11-13:
On Sat, Nov 13, 2021 at 12:22:25AM -0800, emerajid wrote:
> > Yet, there is no developer.
> > if these people existed.
> Do you mean that things are really bad and there are very few (or none
> at all) people who spend their time developing qucs?

It may be good to have an active maintainer to get started. Just not
everyone seems convinced, and nobody has come forward yet. Possible
reason: Qucs is too complex and difficult to maintain. Actually, there
is some sporadic and uncoordinated development happening.

(But this is off topic. The relevant posts, references, discussions and
plans already exist elsewhere.)

> > If you would like to share Qucs/Qucsator binaries, you are not alone.
> but I am not likely to learn programming in the nearest years...
> Especially such hard and complex languages as C++.

I am just trying to state the obvious: If you want something to happen,
do it. Or find/ask somebody capable and willing to help. The best I can
do at the time is answer questions and merge urgent or easy bugfixes.

emerajid, 2021-11-14:
>If you want something to happen, do it

Yes, your are totally right.

>Or find/ask somebody capable and willing to help

This is exactly what I am doing right now. I will still wait some time for occasional luck. Maybe in future something will change. At leasts we must never give up hope. Thank you for answers!

kwmartin, 2022-01-27:
I also had trouble with the install but I was able to bring Qt4 into Ubuntu 20.04 after adding the repository:
sudo add-apt-repository ppa:rock-core/qt4
I think the install command was:
sudo apt-get install qt4-designer
There also seems to be:
sudo apt-get install qt4-dev-tools
I just installed them, but have to re-install qucs.
I am missing components which I need to get up and running.
If I can get it running properly, I will help on the development so it can be used for teaching. It would also be nice if we could get some modern (i.e. 28nm and down) models in, especially something like GF 22-FDX?).

felix-salfelder, 2024-07-02:
Qt4 no longer needed in develop.

closed 2024-07-02 by felix-salfelder