Qucs/qucs #1063: Voltage and Current Sources and Transistors Missing from Fresh Install

kwmartin, 2022-01-27:
Just found out about Qucs, did a fresh install on Ubuntu 20.04. The install was difficult, but I eventually was able to get most of it in except pdftex and repstopdf and some fonts. I was able to run qucs, but on the left pane there aren't any sources or transistors. I'm guessing the libraries were not installed. Can someone help me get them installed so I can netlist a simple circuit. My immediate reaction to the schematic capture is very favorable compared to any other open source schematic capture programs I have found.
My interests are to use it for free courses with young people in India the initial target audience (i.e. my native language is English, India might have lots of interested students); I will also need to something to do figures and equations, I can use Libre Draw, but I really want Qt. I haven't used Qt with C++ (I hate the syntax); I have used PySide extensively, so if I have to use C++ I will. My thoughts are putting together complete courses with each lecture being 5-10 minute U-Tubes, and lots and lots of exercises. I am having extensive discussions with a Prof. from the Univ. of Toronto re getting this project going (he has been working on exercises for well over a year).
I am extremely interested in Verilog-AMS which I believe is absolutely required to be a good analog designer. It appears Verilog-A is available; does anyone know if anyone is working on Verilog-AMS?
If I can get through basics, I will help on the development as qucs may be exactly what I was looking for.
Finally, it looks like it might not be difficult to bring in Xyce as an alternative simulator; any work here?
Finally, finally, can LTI-Spice be run from the command line, this could be another alternative?

kwmartin, 2022-01-27:
Got through the missing sources; no need to help here. Re-installed and found pull-down Combo widget at top of side-bar that gave me acces to sources and transistors.This should be in getting started Tutorial. The transistor symbols don't look bad. The snapping to grid seems to work well.  Now trying to find netlisting command. Schematic capture is usuable except for wire labelling. If I can netlist a hierarchical circuit, I will help on development and will start with changing schematic capture to what I like. Since it's Qt, I will be able to help here despite C++. Any thoughts of coupling in Python as a wrapper above C++? Also, coupling C++ and Julia and C++ and Go are not big deals? It might be nice to freshen Qucs to some newer languages at least at the interface level.

kwmartin, 2022-01-27:
Got a netlist. Totally non-intuitive. You had to include a simulate directive into the schematic and then simulate; yech!. Why not a simple generate netlist under tools menu? Still, open-source, schematic entry is usuable, and it can netlist. It takes forever to compile but I can probably fix this. It is looking more and more that I will get involved in development. I'm bad with git so I might try and run real open loop to get something I like. I've done circuit design for some time so I have strong opinions especially on schematic capture and GUI. Qt generally looks good, to this isn't bad even though it isn't Qt-5. If I get to a workable version I like, I will port to Qt-5. I need to do a few examples to get familiar with it. Is there a list of developers and who is doing what?
Oh, and one of the very first things I'm changing in my version is doing translators from Xml to Yaml and back. Xml is so really not there!

felix-salfelder, 2022-01-27:
On Thu, Jan 27, 2022 at 12:21:55PM -0800, Kenneth Martin wrote:
> [..]
> Any thoughts of coupling in Python as a wrapper
> above C++? Also, coupling C++ and Julia and C++ and Go are not big
> deals? It might be nice to freshen Qucs to some newer languages at
> least at the interface level.

This is off topic here. "Modular Qucs" is work in progress towards most
of these goals. The source is in the "modular" branch, just in case. It
does not fully implement the 0.0.20 features yet, and there is plenty of
work to do...

ra3xdh, 2022-01-27:
> Finally, it looks like it might not be difficult to bring in Xyce as an alternative simulator; any work here?

If you need Xyce/Ngspice as the simulation kernel look at Qucs-S https://ra3xdh.github.io/ There exists an AppImage package, so you need not to compile or install it. This software is stable and ready for operation. Unfortunately I am not able to take part in the project development anymore. 

>  LTI-Spice be run from the command line, this could be another alternative?

LT-Spice was never considered as the simulation kernel, because it is proprietary closed-source Windows-only software.

felix-salfelder, 2022-01-27:
On Thu, Jan 27, 2022 at 12:50:56PM -0800, Kenneth Martin wrote:
> Got a netlist. Totally non-intuitive. You had to include a simulate directive into the schematic and then simulate; yech!. Why not a simple generate netlist under tools menu? Still, open-source, [...]

Modular Qucs has a command line interface for more intuitive work. Qt5
and file formats are also addressed. Tranlators are one thing. Internal
representation of a circuit model or schematic is more relevant.

felix-salfelder, 2022-01-27:
On Thu, Jan 27, 2022 at 01:05:14PM -0800, Vadim Kusnetsov wrote:
> If you need Xyce/Ngspice as the simulation kernel look at Qucs-S
> https://ra3xdh.github.io/ There exists an AppImage package, so you
> need not to compile or install it. This software is stable and [...]

... Modular Qucs has an interface for simulator driver plugins. The Xyce
driver needs to be implemented (but should be very similar to any spice
really). Start with a netlister.

kwmartin, 2022-01-27:
Do we have a bug list we can add to? Or bugs and desired changes?
1) Including Netlist file from File Components is broken.
2) Selecting wire label always selects transistor; Viewdraw way back cycled through nearby components when doing selections which made it much easier select any desired component.
Current Thought: the schematic capture is usable but it and the GUI can really be moved. I don't think there is any other open-source alternative out there. If we could make it into a "Good" schematic capture, and students adopted it, it could become a standard.
Question: can someone direct me in the source code to where the verilog netlister is? Besides teaching, my next interest is doing mixed mode and system-level simulations; that is verilog and verilog-AMS.

felix-salfelder, 2022-01-27:
On Thu, Jan 27, 2022 at 01:24:35PM -0800, Kenneth Martin wrote:
> Question: can someone direct me in the source code to where the
> verilog netlister is? Besides teaching, my next interest is doing
> mixed mode and system-level simulations; that is verilog and
> verilog-AMS.

Still off topic. Please add new tickets on new topics or use the mailing
list for general discussion.

Verilog related plugins (for Modular Qucs) are in plugins/verilog [1].
See tests/ref for example (structural) Verilog output.

[1] https://github.com/Qucs/qucs/tree/modular/plugins/verilog.

kwmartin, 2022-01-28:
Sorry, I'm not a programmer; I'm a circuit designer; so I'm not familiar with proper protocols in git Issues sites. I will try to learn what is acceptable.

felix-salfelder, 2024-07-02:
> I am extremely interested in Verilog-AMS which I believe is absolutely required to be a good analog designer. It appears Verilog-A is available; does anyone know if anyone is working on Verilog-AMS? [...]

Meanwhile we've been working on Verilog-AMS in Gnucap. Ramping up Qucs/GUI to make use of it would need a developper. Please get in touch if you are still interested.

closed 2024-07-02 by felix-salfelder