Qucs/qucs #1064: Qucs and Qucs-S use the same application setting configuration settings

p-brane, 2022-02-12:
Qucs and Qucs-S use the same `File -> Application Settings..` configuration settings/file. Were are the `File -> Application Settings..`  configuration stored? How do you set Qucs to use a specific `File -> Application Settings..`  configuration/file? I would like Qucs and Qucs-S to operate independently of each other.

ra3xdh, 2022-02-12:
Yes, there exists such problem. At the early development stage Qucs-S was intended as a special build of Qucs `spice4qucs` sharing some of its settings. But later the concept of the Qucs-S usage was changed and the settings location was not fixed. The same problem also concerns user libraries, because Spice devices are incompatible with basic Qucs and it shows and error message "Library is corrupt". You may set the Qucs settings home directory at the `Applicationsettings->Locations` tab. But the location of the config file cannot be changed using the GUI. Also use the Qucs-S bugtracker to report the bug on Qucs-S https://github.com/ra3xdh/qucs_s/issues

ra3xdh, 2022-02-12:
Related issue: https://github.com/ra3xdh/qucs_s/issues/71

closed 2022-02-12 by ra3xdh