Qucs/qucs #1067: Setting components values to a variable when using ngspice as Default Simulator. Equation vs Nutmeg

0range0range, 2022-11-06:
Hello,

I have a problem setting value of the component to a variable with `ngspice` as default simulator.

I was trying to follow some 2 year old tutorial and stumbled upon the problem, that I can not give names to the component values.

In the tutorial corresponding directive appear as "Equation": 

[image]

It has "Export" field below. In order to get this I had to switch "Default simulator" to `Qucsator` and then "Add equation".

If on the other hand I use `ngspice` as "Default simulator", when I press on "Add Equation", `Nutmeg` appears, which does not have corresponding "Export" field and I am not sure how to add it: 

[image]

This makes a difference when I try to run a simulation (using `ngspice` as simulator in both cases). Variable definition with "Equation" works. with "Nutmeg" I get an error:

```
Ngspice started...
Netlist line no. 7:
Undefined parameter [rs]
Netlist line no. 7:
Cannot compute substitute
```

**How to set variables with ngspice as "Default simulator"?** or **How to get "Equation" field, without having to reopen the program and changing "Default simulator" to `qucsator` and then back to `ngspice`** If I just define some random variable it works as expected, only when it is substituted as a value of a component do I see a problem.

I do not have any preferences between `Qucsator` or `ngspice`, I just want to follow the tutorial with no problems.

ra3xdh, 2022-11-06:
I need more information what exactly ar you trying to achieve with variables?

>How to set variables with ngspice as "Default simulator"? 

You are using wrong syntax for variables. Units (kOhm, pF) are not supported for both Ngspice and Qucsator. There are two types of variables in Ngspice:

* parametes (`.PARAM`) are computed before the simulation This component is available from `equaiotns` group.
* Nutmeg variables (nutmeg equations) are computed after the simulation and may refer the simulation result. 

This layout follows the Ngspice manual and you may find here how to use the parametes and nutmeg equations https://ngspice.sourceforge.io/docs/ngspice-html-manual/manual.xhtml Nutmeg section knows nothing about parameters. You need to duplicate variables to make it viisible from both sections.

>   How to get "Equation" field, without having to reopen the program

Since v0.0.24 I disabled Qucs equations for new schematics. It is only kept for backward compatibility. There is a lot of incompatible functions (for example S-matrix conversion) between Qucsator and Ngspice equations that could not be converted automatically. Ngspice may produce a cryptic error when trying to use Qucsator function with it. So I decided to exclude Qucs equations to prevent user from insertion it in the schematic. 

Unfortunately, the chapter 4 of the documentation doesn't reflect this change. The usage of the `.PARAM` and nutmeg equations is not described here. Writing documentation is much more time consuming than programming of the application, so don't expect the documentation update in the near future.    

> I do not have any preferences between Qucsator or ngspice

Qucsator has numerous issues with time-domain simulation and not recommended for new desing. Since version 37 Ngspice supports S-parameter simulation and could be used for RF circuit analysis like Qucsator. 

P.S. Don't use this repository for asking questions regarding Qucs-S. Use Qucs-S own repository for reporting issues https://github.com/ra3xdh/qucs_s/issues or asking questions https://github.com/ra3xdh/qucs_s/discussions

ra3xdh, 2022-11-06:
Look at this example. It shows how to use both `.PARAM` and Nutmeg to compute the transient power. 
[image]

0range0range, 2022-11-06:
First of all, thanks a lot!

Duplicating within `.param` works!

> I need more information what exactly are you trying to achieve with variables?

Nothing in particular. I was just following tutorial, then I could not run the simulation presented there and tried to figure out what was wrong.

> You are using wrong syntax for variables. Units (kOhm, pF) are not supported for both Ngspice and Qucsator.

I tried like is shown in the image and it seems to work both when I set `CL=1p` and `CL=1pF` for example. Is `F` simply ignored after `p` is read?

[image]

> P.S. Don't use this repository for asking questions regarding Qucs-S. Use Qucs-S own repository for reporting issues https://github.com/ra3xdh/qucs_s/issues or asking questions https://github.com/ra3xdh/qucs_s/discussions

Okay. Can one move it there somehow? (Sorry, I am not a frequent GitHub user.)

closed 2022-11-06 by 0range0range