Qucs/qucs #1073: python examples?

joamatab, 2023-04-13:
where can we find some python examples for driving Qucs? 

if we can pip install qucs we could make a gdsfactory extension for Qucs

https://github.com/gdsfactory/gdsfactory/issues/681

daquintero, 2023-07-08:
Worth mentioning Pyspice QUCS simulation integration discussion too https://pyspice.fabrice-salvaire.fr/releases/v1.5/simulators.html#qucs

DanielSank, 2024-06-12:
At my job, we've written our own Python driver for Qucs. If it would be useful for us to share that on PyPI I think we can make it happen.

felix-salfelder, 2024-06-12:
> Worth mentioning Pyspice QUCS simulation integration discussion too https://pyspice.fabrice-salvaire.fr/releases/v1.5/simulators.html#qucs

The information on this website is outdated. The gui has been ported to qt5 recently (see develop_filtered branch).

> At my job, we've written our own Python driver for Qucs. If it would be useful for us to share that on PyPI I think we can make it happen.

Thanks. What does it do? I can imagine a python interface for *qucsator*, in that case, the discussion should be moved there.

DanielSank, 2024-06-22:
> Thanks. What does it do?

It reads in netlist files and provides some data structures for overriding values and parsing simulation result files. The workflow is 1) Draw circuit, 2) Read in netlist, 3) Modify netlist with Python code, 4) Invoke qucsator (via Python code), 5) Read simulation file into memory, 5) Do whatever analysis you want in Python.

> I can imagine a python interface for qucsator, in that case, the discussion should be moved there.

Ah yes, that's true,