Qucs/qucs #1074: Incorrect cutoff frequency showing in graph -- sallen key example

csrabak, 2023-04-14:
As a carryover from older installations (the qucs project seems to be issued for qucs 0.0.15) I've a ```Sallen Key Filters_prj``` folder under ~/.qucs which in one of the examples has a "Unity Gain Butterworth Lowpass" circuit.

It seems to be to show a way to find comonent values using equations:
[image]

the table object has been an addition of mine to be able to see the component values arrived at by the simulator.

Whereas the equation for the determination of the resistors (both equal as suggested in several App Notes, like TI's  sloa024b and sloa049d, for example), setting the relationship of the capacitors 1:2, is straighforward, the simulation results show a surprise:
[image]

even if we concede the resistors have been "rounded" the difference in the cuttoff is too far to be considered an error due computation of the exact value, as well as the proposed with 100 points in the simulation does not explain this discrepancy.

I cannot see any issue as the opamp is an "ideal" one for simulation's purpose, and no other features have been put in the model.

However the error is puzzling: 40% error in cuttoff frequency!

The qucs used in this evaluation is 0.0.19 running in a Windows 10 Home, single lang, ver 22H2 (64 bit) machine.

The suggestion in red is from the original file in the project, and the freq in the example was 1 kHz. If so set, the attenuation at this freq is 1.6 dB and the cuttoff is about 1.4 kHz... (which at least is consitent with the 40% errror !!).

felix-salfelder, 2023-04-14:
On Fri, Apr 14, 2023 at 02:16:10PM -0700, Cesar Rabak wrote:
> However the error is puzzling: 40% error in cuttoff frequency!

I can see the cutoff frequency in your example is defined in terms of
the "dB function". This function is probably ill-defined, and could be
the culprit. A recent post [1] somehow relates to it, or at least gives
an idea why "dB" does not work as expected.

If the simulation is wrong in some other way, please provide the file
"netlist.txt" (the qucsator input), and the (raw) simulator output
(some .dat file) pointing out the issue with it.

[1] https://github.com/Qucs/qucs-help/issues/18

csrabak, 2023-04-15:
As I commented there in [1], the definition for the use in this realm is the one universally employed[^2], so the issue is not on this part of the simulation/simulator.

For a Butterworth filter the cuttoff frequency is the one which the amplitude 1/√2 ≈ 0.7071, which is half power or −3 dB. Nowadays for active filters this definition applies for the power in the load of the opamp, not for an "insertion loss" as it is applied for passive filters (and the issue in the mentioned post may arise if the filter is also matching different impedances).

>  please provide the file "netlist.txt"

Here it is: 
```
# Qucs 0.0.19  D:/Users/csrabak/.qucs/Sallen Key Filters_prj/Unity Gain Butterworth Lowpass.sch

OpAmp:OP1 _net0 _net1 _net0 G="1e6" Umax="15 V"
Vac:V1 _net2 gnd U="1 V" f="1 GHz" Phase="0" Theta="0"
VProbe:Pr1 _net0 gnd
R:R1 _net2 _net3 R="R1s" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
.DC:DC1 Temp="26.85" reltol="0.001" abstol="1 pA" vntol="1 uV" saveOPs="no" MaxIter="150" saveAll="no" convHelper="none" Solver="CroutLU"
.AC:AC1 Type="log" Start="1 Hz" Stop="100 kHz" Points="100" Noise="no"
Eqn:Eqn1 gain="dB(Pr1.v)" phase="phase(Pr1.v)" Export="yes"
Eqn:Eqn2 C1s="100e-9" C2s="2*C1s" R1s="1/(2*sqrt(2)*pi*C1s*cutoff)" R2s="R1s" Export="yes"
Eqn:FilterParameters cutoff="2200" Export="yes"
C:C2 _net0 _net3 C="C2s" V=""
C:C1 gnd _net3 C="C1s" V=""
R:R2 _net3 _net1 R="R2s" Temp="26.85" Tc1="0.0" Tc2="0.0" Tnom="26.85"
```
The .dat file generated is attached (GitHub did not like the format so I zipped it).

Notice that the output voltage only drops to 0.707 at about 3kHz, while the equations call for a cuttoff of 2.2kHz:

[image]

So we can for the moment postpone the discussion about the properness of the "dB function", and instead try to see if the issue is elsewhere.

[Unity Gain Butterworth Lowpass.zip](https://github.com/Qucs/qucs/files/11238486/Unity.Gain.Butterworth.Lowpass.zip)

[^2]: Notwithstanding my agreement about this bastardization of the use of decibels.  .  .

csrabak, 2023-04-15:
We found the culprit.

Too concentrated in the issues about the equation generation of the component values. 

We overlooked a glaring mistake which was in plain sight, and it is the schematic itself.

Capacitor C1 ⬆️ is wired to junction of the two resistors while it is supposed to go to R2 side that connects to the non inverting input of the opamp.

Once this is corrected the response comes right.

closed 2023-04-15 by csrabak

felix-salfelder, 2023-04-16:
On Sat, Apr 15, 2023 at 07:44:44AM -0700, Cesar Rabak wrote:
> [..]
> We overlooked a glaring mistake which was in plain sight, and it is the schematic itself.
> [..]
> Once this is corrected the response comes right.

Great, thanks for letting us know.

(It wasn's so glaring after all.)