Qucs/qucs #1075: installation on Windows

mhx-gh, 2023-04-29:
Is qucs_s_win64_1_0_2 compatible with Windows?
The qucs-s.exe binary starts up with tiny icons/text (on a high-dpi display) and the standard way to fix this
( setting the compatibility properties of the exe to "Windows 8" and Change_high_DPI_settings->override high DPI scaling behavior, scaling performed by->System ) results in no application window at all.

ra3xdh, 2023-04-29:
Yes, the Windows binary is tested on physical WIndows regularly. Are you using 4k screen? I have no access to such hardware and cannot test it. Qt library is responsible for application scaling. It should be applied automatically. For Linux the setting of the environment varibale `QT_AUTO_SCREEN_SCALE_FACTOR=1` may help. I cannot suggest an equivalent for Windows. Probably you need to set something inside `qt.conf`` file. 

P.S. you are using wrong repository to report issues. Report issues related to Qucs-S here: https://github.com/ra3xdh/qucs_s/issues Qucs-S is maintained independently from Qucs.

ra3xdh, 2023-04-29:
Closing because the issue reported in the wrong repository. Use https://github.com/ra3xdh/qucs_s/issues to report issues related to Qucs-S.

closed 2023-04-29 by ra3xdh