Qucs/qucs #1076: Qucs-s active filter 1.0.2

mhx-gh, 2023-04-29:
Is Qucs-s active filter 1.0.2 working correctly?

This is the answer for a 5th order lowpass Chebyshev with 3 dB ripple in the passband: 

`Part list`
`Stage#     C1(uF)       C2(uF)      R1(kOhm)      R2(kOhm)      R3(kOhm)` 
`     1     16.077pF    10.000nF     580.224       580.224       290.112`
`     2    273.577pF    10.000nF     221.626       221.626       110.813`
`      3     10.000nF     0.000pF      89.649      1000.000 `  

The above is nowhere near expected. (Why the discrepancy between column header and entry?)
The following values can be found with a websearch: (https://studylib.net/doc/18909792/designing-a-5th-order-low-pass-chebyshev-filter-omar-x.-a...)

`R1 = 20k       C1 = 318n    R2 = 20k      C2 = 5n`
`R3 = 2.112k    C3 = 500p    R4 = 20k      C4 = 312.78n`
`R5 = 20k       C5 = 10nF    R6 = 89.6k    R7 = 8.654k`

With these values I simulated (in another simulator) exactly the correct response?

-marcel

ra3xdh, 2023-04-29:
Yes, the utility operation was tested before releasing and it is correct. The activefilter tool uses different method to calculate the filter circuitry. It follows an approach from D. Johnson, J. Johnson, and H. Moore, A handbook of active filters. Prentice-Hall, Inc, Engewood Cliffs, 1980. This explains the difference. See also here: https://www.researchgate.net/publication/275053492_Qucsactivefilter_-_Active_filter_synthesis_subsystem_of_Qucs

mhx-gh, 2023-04-29:
With the given components and the diagram presented, the Qucs-s filter cuts off at 22Hz with a ripple and dB/oct that looks more like a 3rd order filter. I checked the document you quote and will fill in the values by hand eventually. However, R3 * C2 = 2900us / 1110us (poles at ~50Hz / 125Hz) of the first and second stage looks a bit strange.

csrabak, 2023-06-12:
The document pointed to in April 29th reply by Vadin, is a thorough review of the theory for active filter design, but falls short of being useful for the use of the utility. . .

If one sellects the BPF the GUI comes as:
[image]

However, in no page of 41 pages report (or the help system) it is found the meaning for the "transient bandwidth", and once Butterworth response is chosen it is at least confusing still have a "bandpass ripple" field showing as to be filled in...

I also didn't find in the accompanying text why the order of the filter cannot be chosen instead. . .

I suggest an addendum be written to the text (or an another additional manual) showing examples of the use of GUI.

Regards.

csrabak, 2023-06-13:
I got a hand in the referred book, and I think I found the issue. It may be a matter of the famous joke about translations, in the Qucs acive filter document (reacheable via the Qucs menu, and also pointed to a ResearchGate page) the Transition Width (Sec 5-3 of the book) of the two transition bands was referred to as _transient_ width. 

In filter design transient is associated to the step response and the measures of overshoot, etc. So I think is confusing to mantain this naming, be it in the Qucs manual, be it in the GUI.

On the other hand, the dimension lines for the TW in the figure is shown above the point in the roll-off curve where the attenuation for the stop frequency is specified, As in the above figure.

It is interesting to note as well that although the idea of transition width is used in the book for all the response types, in the GUI only for the band-pass it appears!

HTH

mhx-gh, 2023-07-29:
One problem I had is:
```
Stage#     C1(uF)       C2(uF)      R1(kOhm)      R2(kOhm)      R3(kOhm)
     1     16.077pF    10.000nF     580.224       580.224       290.112
     2    273.577pF    10.000nF     221.626       221.626       110.813
     3     10.000nF     0.000pF      89.649      1000.000
```
Is C1 here 16.077 pF, 16.077uF, or 16.077e-12*1e-6 Farad? (See column header). Is R3 290.112k, or 290.112 Ohms?
With this confusion, there are already 6 combinations to try out.

Did you get the response you intended, when simulating the final filter with Qucs / any simulator?

-marcel

mhx-gh, 2023-07-29:
One problem I had is:

Stage#     C1(uF)       C2(uF)      R1(kOhm)      R2(kOhm)      R3(kOhm)
     1     16.077pF    10.000nF     580.224       580.224       290.112
     2    273.577pF    10.000nF     221.626       221.626       110.813
     3     10.000nF     0.000pF      89.649      1000.000
Is C1 here 16.077 pF, 16.077uF, or 16.077e-12*1e-6 Farad? (See column header). Is R3 290.112k, or 290.112 Ohms?
With this confusion, there are already 6 combinations to try out.

Did you get the response you intended, when simulating the final filter with Qucs / any simulator?

-marcel

ra3xdh, 2024-07-02:
Fixed in https://github.com/ra3xdh/qucs_s/issues/398 Closing because reported in Qucs repository instead of Qucs-S.