Qucs/qucs #1082: support SAX simulation

joamatab, 2024-01-27:
How could we enable [SAX](https://flaport.github.io/sax/) simulations from QUCS?

@flaport
@nikosavola

[image]

ra3xdh, 2024-01-28:
You have used wrong repository to report issues for Qucs-S. Please report issues here in the future: https://github.com/ra3xdh/qucs_s/

The adding of new simulation backend in Qucs-S should be technically possible. But adding SAX support may require a significant effort. The netlist format of SAX differs from SPICE. Also it is required to implement a dataset parser to convert dataset produced by SAX to the Qucs internal XML dataset. I don't consider this task to be implemented in the near future, unless someone will provide a patch. 

I am closing this because it is reported in a wrong repository. I would qualify this request a discussion, and you may continue here: https://github.com/ra3xdh/qucs_s/discussions

felix-salfelder, 2024-01-28:
This request is an interesting one, because it hints at the original purpose of the QUCS UI.

We have work in progress on a Qucs sucessor, "Modular Qucs". It is designed to support different file formats and simulator backends in terms of run time extensions. Some early support for Verilog netlists and schematics is available, supplementing the limited "sch" and "txt" file formats.

Modular Qucs will need a revival with the advent of a Verilog-AMS simulator. Once this has happened, any other file format or simulator will be a relatively trivial addition. An alternative option will be be to support Verilog in SAX, because it is a well accepted standard that unifies previous attempts.

(I do not see a reason to close this, even if it will take time. Feel free to discuss.)

ra3xdh, 2024-01-28:
> We have work in progress on a Qucs sucessor, "Modular Qucs".
> I do not see a reason to close this, even if it will take time.

The last commit on a `modular` branch was three years ago. Any progress since this time?

felix-salfelder, 2024-01-28:
On Sun, Jan 28, 2024 at 10:46:45AM -0800, Vadim Kuznetsov wrote:
> > We have work in progress on a Qucs sucessor, "Modular Qucs".
> > I do not see a reason to close this, even if it will take time.
> 
> The last commit on a `modular` branch was three years ago. Any
> progress since this time?

Progress is a matter of perspective. Realistically, I got dragged into
more foundational work.

We are about to conclude the first round on Verilog-AMS [1]. One of the
deliverables, a "replacement for ADMS", is ready. Its based on modelgen,
the model compiler from Gnucap. Complementary, Gnucap now implements
some essential concepts defined in Verilog-AMS, still with an analog
bias. As a side effect, the complete infrastructure needed underneath
the Modular Qucs has become tangible. As a shared library, it's not
supposed to be committed here.

Our next project [2] will continue on that path adding discrete
modelling and mixed simulation of large circuits. Towards the topic of
this discussion, we will document the details of a standard data
interchange format for schematic and layout, "to finally leave behind
the long painful era of vendor specific netlists", c.f. [3]. A Modular
Qucs revival might become an early adopter, as it's close. Clearly, any
other will do just as well.

The invitation to join the initiative remains open, and a bunch of
new opportunities have emerged...

cheers
felix

[1] https://nlnet.nl/project/Gnucap-VerilogAMS/
[2] https://nlnet.nl/project/Gnucap-MixedSignals/
[3] https://github.com/pascalkuthe/OpenVAF/issues/42

closed 2024-07-02 by felix-salfelder