Qucs/qucs #1083: S-parameter file causes simulation error

zodoczi, 2024-03-01:
Hi, 

I'm playing with this RF switch, where quite hopeless what should be wrong with my simulation as many different switches was working before. I copy the simulation error log also attach the S parameter file here. In a text editor I was not finding any strange things however sure there is something not usual here. Thats sure that the S parameter file was made with a real desktop VNA.

`Errors and Warnings:
--------------------

line 23: no trailing end-of-line found, continuing...
line 23: syntax error, unexpected Float, expecting $end
checker error, found 6 options
checker error, option `hz' occurred 2x
checker error, option `s' occurred 2x
checker error, option `ri' occurred 2x
checker error, option `hz' occurred 2x
checker error, option `s' occurred 2x
checker error, option `ri' occurred 2x
[D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A.s5p.zip](https://github.com/Qucs/qucs/files/14460361/D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A.s5p.zip)
`

Thank you very much!

zodoczi, 2024-03-01:
Okay, looks like AWR S data repair tool magically solved this :) Now my simluation runs smoothly with the repaired S-parameter!
https://awrcorp.com/download/kb.aspx?file=/Questions/SdataConversion.zip

[D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A_mwo.s5p.zip](https://github.com/Qucs/qucs/files/14461616/D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A_mwo.s5p.zip)

zodoczi, 2024-03-01:
BTW any idea while de original S parameter file was failing? 

Here is the repair process, some issues shown about negative character used as delimiter

[image]

Thanks

felix-salfelder, 2024-03-01:
On Fri, Mar 01, 2024 at 05:31:56AM -0800, Zoltan Doczi wrote:
> BTW any idea while de original S parameter file was failing? 

Could you provide a netlist.txt please?

thanks

zodoczi, 2024-03-02:
> On Fri, Mar 01, 2024 at 05:31:56AM -0800, Zoltan Doczi wrote: BTW any idea while de original S parameter file was failing?
> Could you provide a netlist.txt please? thanks

Attaching the whole simulation, and the working (D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A_mwo.s5p) and the problematic (D_PE42540_SOLDERED_25C_3.300V_3.300V_0.000V_RF1_S5P_SN22_A.s5p) parameters.

Now the upper s parameter box (with the ports attached) is the corrected s parameter file (_mwo ending) the lower box contains the original but bogus s parameter file. Once you disable the lower box the simulation will run without any problem.

Thank you
[sparameter_freeze.zip](https://github.com/Qucs/qucs/files/14468651/sparameter_freeze.zip)

felix-salfelder, 2024-03-02:
On Sat, Mar 02, 2024 at 12:59:56AM -0800, Zoltan Doczi wrote:
> > On Fri, Mar 01, 2024 at 05:31:56AM -0800, Zoltan Doczi wrote: BTW any idea while de original S parameter file was failing?
> > Could you provide a netlist.txt please? thanks
> 
> [sparameter_freeze.zip](https://github.com/Qucs/qucs/files/14468651/sparameter_freeze.zip)

There is no netlist.txt in the archive (?).

zodoczi, 2024-03-02:
Sorry, new to Qucs have no idea how to export the netlist. I've just built up the simulation in the GUI interface, thanks

felix-salfelder, 2024-03-02:
On Sat, Mar 02, 2024 at 01:56:39AM -0800, Zoltan Doczi wrote:
> Sorry, new to Qucs have no idea how to export the netlist. I've just built up the simulation in the GUI interface, thanks

Its in ~/.qucs/netlist.txt, after the simulation attempt.

thanks

zodoczi, 2024-03-02:
Ouch, sorry for the confusion but I see nothing like that in my folder :( I'm using it in Windows 11, should I try it under Ubuntu?

zodoczi, 2024-03-02:
okay found it, on windows its stored in a completely different folder, under users, find attached
[netlist.zip](https://github.com/Qucs/qucs/files/14469146/netlist.zip)
Thanks

felix-salfelder, 2024-03-02:
On Sat, Mar 02, 2024 at 03:37:28AM -0800, Zoltan Doczi wrote:
> okay found it, on windows its stored in a completely different folder, under users, find attached
> [netlist.zip](https://github.com/Qucs/qucs/files/14469146/netlist.zip)

Okay, I can reproduce this.

Qucsator SPFile has a limit on the number of characters or numbers in
each line. the _mwo version has more line breaks...

I'm working on successor/substitute for qucsator once in a while. May I
use your netlist as a test case?

(will move this issue to qucsator.)

Best wishes
felix