Qucs/qucs #1085: How to build/install this software

DanielSank, 2024-06-23:
My work group in my company is using Qucs through a half-baked app image that we cobbled together. We are expanding our use of circuit simulation tools and so we'd like to get up to date with the latest Qt version etc.  The build instructions mention a tarball. I have a few questions:

1. Where is this tarball and/or are the build instructions still corect?
2. The sourceforge page was last updated in 2017, but the Git repo was last updated 2 weeks ago. Why is there such a disparity there?

ra3xdh, 2024-06-23:
@DanielSank You may look at the Qucs-S https://github.com/ra3xdh/qucs_s/ It is ported to Qt6, supports both Qucsator and Ngspice, and has support for many new features from the latest Ngspice releases.

felix-salfelder, 2024-06-23:
On Sun, Jun 23, 2024 at 09:20:17AM -0700, Daniel Sank wrote:
> My work group in my company is using Qucs through a half-baked app
> image that we cobbled together. We are expanding our use of circuit
> simulation tools and so we'd like to get up to date with the [...]

The tarball referred to in the documentation could be the one shared by
former maintainers. It used to be created by "make dist".

Note that I have "released" 0.0.20 for archival purposes only. I am not
able to build the old gui, nor am I willing to spend much time on
archeology.

What we have is a development version on top of 0.0.20. Documentation
updates have not been addressed yet. Nor any form of deployment.

> 2. The sourceforge page was last updated in 2017, but the Git repo was
> last updated 2 weeks ago. Why is there such a disparity there?

Similar reasons. Some long time ago, Qucs has moved half way to GH, and
today it is all over the place.

To make sense of a future Qucs, the schematic file format needs to
change to address the issue of interoperability.

Hope this answers your questions, please get in touch if you'd like to
help.

Best wishes
felix

DanielSank, 2024-06-24:
@ra3xdh, thank you. Yes I know about your project and I have a few questions about it, which I will ask in the `qucs-s` repo.

@felix-salfelder, thank you for the contextual information.

> I am not able to build the old gui, nor am I willing to spend much time
> on archeology.

To be clear, I do not think my work group is interested in archaeology either. Our interest would be finding a circuit simulation tool which is reasonably well maintained such that we can us it, and contribute a few features to support our own work.

> What we have is a development version on top of 0.0.20. Documentation
> updates have not been addressed yet. Nor any form of deployment.

How would I build that?

> To make sense of a future Qucs, the schematic file format needs to
> change to address the issue of interoperability.

Is there an issue about that? Searching "schematic" in the open issues, I do not see anything related except perhaps #943 but that looks like expected behavior.

felix-salfelder, 2024-06-24:
On Sun, Jun 23, 2024 at 11:06:08PM -0700, Daniel Sank wrote:
> > What we have is a development version on top of 0.0.20. Documentation
> > updates have not been addressed yet. Nor any form of deployment.
> 
> How would I build that?

I made a quick update to the "develop_filtered" branch README. Basically
you can build it either with autotools or with cmake, subject to issues
you might find.

> > To make sense of a future Qucs, the schematic file format needs to
> > change to address the issue of interoperability.
> 
> Is there an issue about that? Searching "schematic" in the open
> issues, I do not see anything related except perhaps #943 but that
> looks like expected behavior.

This is quite new. I found funding for Verilog-AMS, and one of our tasks
is to define a Verilog based schematic interchange file format. The file
format spec will be published when it's there. It's not for Qucs, and
whether Qucs will use it is up in the air.

The closest out there is Verilog schematics in modular Qucs. It had a
bunch of issues though. Test cases in the "modular" branch give an idea
of the direction. What is missing there is the use of attributes, and a
document describing it.

Best wishes
felix

felix-salfelder, 2024-07-02:
(README updated accordingly)

closed 2024-07-02 by felix-salfelder