Qucs/qucs #1088: If there is a new release, please pull the new version of ASCO too

buenoshun, 2024-07-08:
When there is a new QUCS release, then please pull the latest version of ASCO.
ASCO was updated in 2022, they have already fixed a bug (with my help) when working with QUCS simulations. The QUCS simulation crashed when using the optimizer sim controller, but after some emailing and a new release it now works. Currently users have to manually download ASCO and copy it into the QUCS install folder to replace the one supplied with the QUCS installation.
This should be easy to do.

felix-salfelder, 2024-07-08:
Qucs (this repo, this software, this package) won't ship any third party tools in the future. The former meta-package has moved to the qucs-pkg repo. To survive, it will need an update and a maintainer.

Please also consider conventional channels for packageing such as this [1].

[1] https://aur.archlinux.org/packages/asco

closed 2024-07-08 by felix-salfelder

buenoshun, 2024-07-18:
If ASCO is not shipped with QUCS, then what happens when someone places the Optimization simulation controller into a schematic and runs it? It will run qucsator instead, or we have to manually install ASCO, or the sim controller will be removed from the QUCS GUI? What am I missing?