Qucs/qucs #1103: Distro packaging support

yarda, 2025-03-25:
Hi,

I am packaging qucs for Fedora. The following are problems we have encountered with 0.0.20 qucs release. If we agree on solution I am OK to open PR(s) where appropriate.

# On Fedora 64 bit it installs library to /usr/lib not /usr/lib64
I think it may use cmake `GNUInstallDirs` and not hardcode `/lib`, we used the following patch in Fedora:
```
--- a/qucs/CMakeLists.txt
+++ b/qucs/CMakeLists.txt
@@ -330,16 +330,16 @@ endif(WIN32)
 #
 install(
   TARGETS qucs
-  BUNDLE DESTINATION bin COMPONENT Runtime
-  RUNTIME DESTINATION bin COMPONENT Runtime)
+  BUNDLE DESTINATION ${CMAKE_INSTALL_BINDIR} COMPONENT Runtime
+  RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR} COMPONENT Runtime)
.
 # set Windows runtime location for libqucschematic See:
 # http://www.cmake.org/pipermail/cmake/2010-June/037461.html
 install(
   TARGETS qucsschematic
-  RUNTIME DESTINATION bin COMPONENT runtime
-  ARCHIVE DESTINATION lib COMPONENT devel
-  LIBRARY DESTINATION lib COMPONENT library)
+  RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR} COMPONENT runtime
+  ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR} COMPONENT devel
+  LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR} COMPONENT library)
.
 #
 # Install needed Qt plugins by copying directories from the qt installation One

```
Maybe the `DESTINATION` could be fully dropped (or replaced by`TYPE`) and the cmake defaults used?

# It installs libqucsschematic without SONAME
If this library is meant to be private it should be installed outside of the linker path (e.g. `/usr/lib(64)/qucs/`). If this library is public (e.g. other projects can link against it) it should have SONAME. In case it's unversioned, it should still use some dummy SONAME like e.g. 0.0. I don't know whether the library is public, but it is installed as a public, so in Fedora we used the following patch:
```
--- a/qucs/CMakeLists.txt
+++ b/qucs/CMakeLists.txt
@@ -24,6 +24,9 @@ set(PROJECT_VENDOR "Qucs team. This program is licensed under the GNU GPL")
 set(PROJECT_COPYRIGHT_YEAR "2014")
 set(PROJECT_DOMAIN_FIRST "qucs")
 set(PROJECT_DOMAIN_SECOND "org")
+set(qucsschematic_VERSION_MAJOR 0)
+set(qucsschematic_VERSION_MINOR 0)
+
.
 #
 set(CMAKE_BUILD_TYPE Debug)
@@ -294,6 +297,9 @@ add_library(
.
 # link system libraries
 target_link_libraries(qucsschematic ${QT_LIBRARIES} ${CMAKE_DL_LIBS})
+set_target_properties(qucsschematic PROPERTIES SOVERSION "${qucsschematic_VERSION_MAJOR}")
+set_target_properties(qucsschematic PROPERTIES VERSION "${qucsschematic_VERSION_MAJOR}.${qucsschematic_VERSION_MINOR}")
+
.
 #
 # CMake's way of creating an executable
```

# It doesn't install headers
If the library is public, there should be headers installed for it. It seems there are some headers included in the release archive, but I don't know whether there are meant to be used together with the library. To be safe we packaged it to the devel package in Fedora:
```
--- a/qucs/CMakeLists.txt
+++ b/qucs/CMakeLists.txt
@@ -27,6 +27,7 @@ set(PROJECT_DOMAIN_SECOND "org")
 set(qucsschematic_VERSION_MAJOR 0)
 set(qucsschematic_VERSION_MINOR 0)
.
+include(GNUInstallDirs)
.
 #
 set(CMAKE_BUILD_TYPE Debug)
@@ -295,6 +296,8 @@ add_library(
   $<TARGET_OBJECTS:diagrams> $<TARGET_OBJECTS:dialogs>
   $<TARGET_OBJECTS:paintings>)
.
+target_sources(qucsschematic PUBLIC FILE_SET HEADERS FILES ${QUCS_HDRS})
+
 # link system libraries
 target_link_libraries(qucsschematic ${QT_LIBRARIES} ${CMAKE_DL_LIBS})
 set_target_properties(qucsschematic PROPERTIES SOVERSION "${qucsschematic_VERSION_MAJOR}")
@@ -339,7 +342,8 @@ install(
   TARGETS qucsschematic
   RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR} COMPONENT runtime
   ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR} COMPONENT devel
-  LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR} COMPONENT library)
+  LIBRARY DESTINATION ${CMAKE_INSTALL_LIBDIR} COMPONENT library
+  FILE_SET HEADERS DESTINATION ${CMAKE_INSTALL_INCLUDEDIR}/qucs COMPONENT devel)
.
 #
 # Install needed Qt plugins by copying directories from the qt installation One

```

yarda, 2025-03-25:
Thanks for the info.

So the recommended way is to use autotools for building?

Can we package the git snapshot for Fedora stable release? Is it usable enough?

felix-salfelder, 2025-03-25:
On Tue, Mar 25, 2025 at 07:32:06AM -0700, Jaroslav Škarvada wrote:
> So the recommended way is to use autotools for building?

Can't recommend either. I use autotools, and I can review fixes to it.
If you know how to fix CMake, go for it. In a perfect world, you'd get
identical results.

> Can we package the git snapshot for Fedora stable release? Is it usable enough?

Once there is a snapshot, it will hopefully be more usable as the
previous releases. There hasn't been a release due to the Qt4 stalemate
for quite some time. See [0] for details.

Right now, it might make sense to follow git/develop, similarly to the
way Arch/AUR does [1]. Not sure if/how Fedora supports this.

[0] https://sourceforge.net/p/qucs/news/2025/01/post-0020-trajectory/
[1] https://aur.archlinux.org/packages/qucs-git

yarda, 2025-03-25:
OK, thanks for the info. We previously used autotools when building the RC release, so no problem with it.

I will try to package the git snapshot and we will see.  In Fedora we probably cannot do nightly snapshots, but updating time to time shouldn't be a problem.

Anyway, closing.

felix-salfelder, 2025-03-25:
On Tue, Mar 25, 2025 at 10:42:47AM -0700, Jaroslav Škarvada wrote:
> OK, thanks for the info. We previously used autotools when building the RC release, so no problem with it.

Perfect. Thanks for the effort.