Qucs/qucs #1104: Gui is not the best + no qt6

Rayan-Haddad, 2025-05-04:
I guess that the apps GUI is not the best currently, i can't do a lot about it since i am busy with other things but i guess i will solve this when i am free, anyways, i guess that moving to GTK 4 is better then using c++ and Qt since GTK 4 is a more open library to other languages and its totally open source, so i guess that this would help

felix-salfelder, 2025-05-05:
Thanks for offering help, both the ROADMAP and CONTRIBUTE files may give an idea about what is going on, and how to help.

Qucs uses Qt (much more than necessary). There is work in progress cleaning up the code, also decoupling data from the GUI. We do not plan to replace the Qt, but without prior cleanup, a GTK port will likely lead to a rewrite.

The highest priority right now is to implement a standard file format which will enable sharing circuits across different tools. One of those tools could be a schematic editor built with GTK. It looks like a distraction to me, but feel free to try.

Rayan-Haddad, 2025-05-05:
Thanks, my main goal at the time is to improve the GUI, if it seems to hard to move to GTK, i guess i will do some good GUI work with QT, I want free and opensource software like Qucs to defeat payed software and one of the main differences between opensource and paid software is the GUI, in Paid software GUI feel much better and you easier to use and much modern, and that's what i want to reach with Qucs, a good modern easy to use GUI, but when i gain the necessary knowledge to help on other things i will of course

Rmano, 2025-05-06:
Hi all! 

**Disclaimer**: I have lost contact (more than a bit) with the development (you know, RL and an appointment as Head of Dept...) after publishing the [AppImage of 0.0.20-rc2](https://github.com/Rmano/qucsAppImagesBuild/releases/tag/v0.1) to continue using QUCS on my computer when QT3 (or it was 4?) was removed from everywhere, so I do not know if what I am saying here has been discussed before --- so please bear with me.

For a couple of years, I have been using `qucsator` models on qucs-s. It works perfectly, the interfaces have been ported to Qt6, and the old projects simply work. 

Yep, it has the added complexity of having more simulation engines (but hey, sometimes NG-spice is useful, especially to use manufacturer-provided models that sometimes *do* work), and the documentation is, well, below par (but actively worked on). 

 The interface still is not the best in town (I really think that the fact that the projects are all in one directory and they can't be simply scattered around in your file structure is a pain in the back, but well, I seem to be almost alone in that).

The difficult question is: wouldn't it be better to join forces here?  I am trying to help the documentation effort on qucs-s, and suggesting things, and the development is going on at quite a pace there. The interface has already been ported to modern Qt, so the efforts could be oriented to grow from there... Or at least there could be cross-fertilization, by using the changes they have done here.

felix-salfelder, 2025-05-06:
> The difficult question is: wouldn't it be better to join forces here? I am trying to help the documentation effort on qucs-s, and suggesting things, and the development is going on at quite a pace [...]

Hi Rmano.

Thanks for packaging Qucs-0.0.20. Let me use this opportunity to express my regrets regarding Qucs-S. A better outcome appeared to be tangible but no funding has emerged. I ended up working on a higher priority project when Verilog-AMS turned out to be feasible.

I can see the emphasis on Spice in Qucs-s. There has been some progress on making it look shiny indeed, to the point where users accept the lock-in. Qucs has been in a bad shape since recently, but is now growing the support for an industry standard, the opposite of lock-in. The current development is overdue, considering how the world has moved on.

- We now support IEEE1364 based schematic files (in develop) and soon will have a suitable component library.
  Qucs-s still uses the legacy schematic format, which is useless outside of Qucs or Qucs-s.
- We are moving beyond Spice in terms of modelling capabiities and supported simulation engines.
  Icarus-Verilog and Gnucap read the new schematics. Gnucap offers a free (and growing) Verilog-AMS simulator.
  (Use a commercial simulator if you lack the ethics and have the financial background.)
- I have severe doubts regarding your claim about "modern Qt in Qucs-S". I ported the Qt3 wrapper from Qucs-S myself.
  We got rid of it over a year ago. More steps are needed, but you need to know the direction. Please read the code.
- Sometimes ngSpice is useful (emphasis on "sometimes"), and so I have already implemented initial support for it.
  (In a perfect world, the ngSpice team would take care of Verilog-A, rendering my attempts useless.)
- We have expert advice, run on NLnet funding and offer paid internships.
  Still need to get universities back on board, so spread the word.

Regarding cross-fertilisation. Sure, I took the Qt3 wrapper from Qucs-S as a band aid. It was totally useful. Yet, when I recently nudged the Qucs-S folks regarding standard file formats, the response was "Verilog-AMS is not a standard", contrary to all evidence. I am not sure what to make of that.

Eventually, it would be nice to see Qucs-S adopt a [standard file format](http://gnucap.org/dokuwiki/doku.php/gnucap:user:netlist_import_and_export:qucs) in the long run. Without data exchange, every FOSS/EDA project will have to implement all of EDA. Options are to either do something about it or watch them fail.

Rmano, 2025-05-06:
Hi — ok, I will comment (I feel obliged) because I started the discussion. Sorry if I stirred up some old problems I did not know.

About the interface of both qucs and qucs-s: I really find the only problematic thing, for my workflow, the fact that there is a project directory. I'd like to have the simulation scattered in my file tree, for example, in the same directory as my class notes and exercises (like exam solutions). I know I *can* do it, but it's not really straightforward.  One of the most used programs around here is LTspice, and I find that *everything* there is worse than Qucs/Qucs-A as a matter of interface. Don't even get me started about graphics.

I know nothing about Verilog, so I'll not comment here. But I think it is true that each and every program I used (qucs/qucs-s/ltspice/Kicad and, in my remote past (around 1998), Cadence) uses a different format for schematic. I'd love to be able to feed it my `circuitikz` diagram, but that is for the next life (I am joking). The nice thing with Qucs-S now is that I can, for example, download from Texas Instruments the model of, say, the INA826, and it works. 

So let's agree to disagree and continue to live happily.  I will continue to try to help when I can, both projects, although I have to say that the task of helping with Qucs-S doc is the one more "reachable" for me at this moment...

Sorry again for creating the discussion.

ra3xdh, 2025-05-06:
> for my workflow, the fact that there is a project directory. 

It is possible to fix. You may open a request in Qucs-S repository. Furthermore the both Qucs-S and Qucs support schematics without projects. I prefer to use schematic without projects. 

> or example, download from Texas Instruments the model of, say, the INA826, and it works.

Yes, it is one of the key features of every circuit simulation software from the user perspective. Simulator which has no seamless support of third party models  will have only limited application.

felix-salfelder, 2025-05-06:
On Tue, May 06, 2025 at 08:56:39AM -0700, Vadim Kuznetsov wrote:
> At the first point I don't consider a merging of Qucs and Qucs-S.

Thanks Vadim.

Whether you consider to add a Verilog-S support to Qucs-S is up to you.
There is certainly no need to merge Qucs and Qucs-S.

If the Qucs-S schematic editor has improved (as you say), we might port
it back to Qucs. But first things first.

> [..] The majority of existing circuit simulation software are using
> unique schematic formats and users are happy.

This looks empirically true but is misleading [1]. Sharing files across
schematic editors and simulators is quite a recent accomplishment and
still under development. Sharing files has shown to be very fruitful in
the digital-only domain, see Verilog-95.

> The simulator should provide PSpice/LTspice syntax compatibility

Analog/mixed circuit simulators relevant today support Verilog-A or AMS.
Sure, you can always pick on Gnucap lagging behind. Yet there is little
to no Verilog-A support in any of the other FOSS simulators, though
Verilog-A and AMS is just the start. System-Verilog and System-Verilog-
AMS are lurking. User happiness is nice to have, but innovation is due.

Full PSpice/LTspice compatibility is a noble goal. But we need to be
able to turn user data into something that generally works in modern
simulators [2]. In more than your favourite one. What if Spice3 is
suddenly found to be insufficient [3]?

> I will consider the adding its support if the mass migration to the
> new schematic format will begin.

You cannot take the lead by following. Not by catching up with
commercial Spice dialects, and not by waiting for mass migration.

[1] https://en.wikipedia.org/wiki/Selection_bias
[2] http://gnucap.org/dokuwiki/doku.php/gnucap:manual:tech:spice2verilog
[3] https://web.archive.org/web/20060926034314/http://www.cs.sandia.gov/nacdm/talks/Nagal_Larry_NACDM2004.pdf

felix-salfelder, 2025-05-06:
> Sorry again for creating the discussion.

Thanks for chiming in, Rmano. I did not mean to discuss Qucs-S here, but anyway, it might have been overdue.

It turns out, all of us still have their individual goals, and forking projects is not an unusual way out. In Qucs, we have a ROADMAP.. It would be nice to have one in Qucs-S, too, so potential contributors can choose more easily.

ra3xdh, 2025-05-06:
> If the Qucs-S schematic editor has improved (as you say), we might port
it back to Qucs. 

You may try. GPL license allows code exchange between GPL projects. This works in both directions. 

>  It would be nice to have one in Qucs-S, too, so potential contributors can choose more easily

See here: https://github.com/ra3xdh/qucs_s/wiki/Roadmap

felix-salfelder, 2025-05-07:
Thanks Vadim.

The ROADMAP now covers more GUI related aspects of Qucs.

Rayan-Haddad: I'm closing this, feel free to open a new issue on anything particular.

closed 2025-05-07 by felix-salfelder