Qucs/qucs #1105: Qucs version 25.1.2

Cy-27, 2025-05-07:
​Issue Description:​​
When opening a SPICE file in the SPICE library device, the file type filter incorrectly displays ​​"+.ckt"​​ instead of the standard wildcard format ​​"*.ckt"​​. This prevents users from properly locating or selecting .ckt files.

​​Observed Behavior:​​
The application shows +.ckt in the file filter dialog, which is non-standard and unexpected.

ra3xdh, 2025-05-07:
Closing because reported in a wrong repository. Qucs is not related to Qucs-S project. Report issues here: https://github.com/ra3xdh/qucs_s/issues See https://github.com/ra3xdh/qucs_s/issues/1349 to track the status.

closed 2025-05-07 by ra3xdh

felix-salfelder, 2025-05-07:
There is no Qucs version 25.1.2. Qucs-S maybe?